drunsinn/pyLSV2 #6: Program stalling

WouterElfrink, 2021-02-02:
Hey Max, 

I've got a question, when i'm connected to the machine and a program is running all is going ok. But when a sub-program is called the machine stalls. I can read that another program is called and the machine puts the tool away but after that it stalls.

The controls on the TNC don't work as well when this is happening

If I disconnect from the machine, the machine continues and works fine after that.

Do you know how this is happening ? Could it have something to do with a tool change or state change?

Thanks in advance!
Wouter

drunsinn, 2021-02-02:
Hello @WouterElfrink,

I haven't used theses function in regular use, only the file transfer stuff. I have never measured if the transfer times differ if there is a program running or not but I have not noticed anything in particular.

Just to be sure I understand you correctly: Does the control stall the machining or the python program?

Since you traced it to the blocking mode the control seems to be prioritizing machining over responding to LSV2 commands. How often do you request an update? Maybe reducing the frequency could reduce the risk of the machine stalling.

The blocking mode was necessary to make sure to the bigger telegrams arrive in one pice and not split into separate parts. I can look into making it work without it but no guarantee in success.

Another option would be for you to move the communication into a separate task. That way the rest of the python application would be responsive while pyLSV2 is waiting for the response.

I hope these recommendations help you, again I have no experience with reading plc memory and status information repeatedly.

WouterElfrink, 2021-02-02:
Hey @drunsinn,

The control (and machine) stalls, the python program just keeps running.

I did some testing and found out that it has something to do with the `get_program_stack` and/or `get_program_status`. 

When I executed these statements, after connecting with `con.connect` the machine would stall when it wanted to change tools. When I ran the same script without executing these statements it worked fine.

Also when running the script without the statements mentioned above, the plc marker reader worked fine as well. 

I request an update regular, now every 3 seconds. This could be reduced indeed. 

I now wait until the machine gives back a M02 code with the respecitve Marker, but the time that this Marker is sent is very short. I tested it without an time interval and then it notices the change. But when I use a time interval, because the marker time is so short, it isn't noticed.

Thanks for the recommendations! I'll try to work with them.

Wouter

drunsinn, 2021-02-06:
Hello @WouterElfrink 

Thanks for the update!

I dont think it has anything todo with the socket being full. The configuration step we had problems in the earlier issue negotiates the buffer size supported by the control. This size is used to get the absolute maximum number of bytes received in one telegram. Telegrams like the one used to transmit files use a special response code to indicate that additional telegrams are expected. A telegram can be split over multiple TCP packets which is why i decided to use this mode for the socket.
pyLSV2 is the first project for me that uses any low level network stuff so there are most certainly ways to improve performance and stability. If you have something feel free to let me know and/or send me a pull request!

The functions for reading the status and the program stack aren't that much different from reading the version information. The only difference is that the information always changes while the other one stays the same for each control.

Since the machine slows down it could be a number of things going on. This is just speculation but maybe it is some kind of protection against interferences during critical operations. Since LSV2 dosen't really have any protection and the PLC key number is also quite well known HDH could use some other ways to stop us from using certain functions without paying for the DNC license.

To be honest, i am more surprised that it works as good as it dose without it messing up the control in some way 😅 That's also why I chose the MIT License since i can't guarantee anything in regards to it working or not.

I will leave this issue open for now as maybe there are infact ways to improve the situation.

WouterElfrink, 2021-02-23:
Hey Max, 

Just wanted to let you know that I found a way around it.

I connect to the machine to read the current program and some lines from the NC-file, then I disconnect from the machine. I reconnect to the machine and waits until the M02 marker changes from 0 to 1. Then execute some statements.

This way works for me. Now I need to figure out how to do that with multiple machines at the same time. Any tip that you would have is welcome!

Thanks for all the help anyway !!

Wouter

drunsinn, 2021-02-23:
Hello Wouter,

that's great news! Good that you found a workaround. 👍
Its interesting behavior but who knows what the control is doing with the connection while it is idle. Connecting and Disconnecting seems to be easier on the control than leaving it open....

I think the solution for multiple connections depends on what you want to do. My application for LSV2 is to transfer files, for that I use threads. Each thread creates it's own LSV2 object and connects to one machine. After the transfer is finished the thread returns a status flag if it was successful or not. This way each thread has it's own socket and there is no problem with keeping track of resources or the thread blocking each other.
This works since I don't have to do anything while the files are being transferred. The main loop waits until every thread is finished bevor continuing.

Do you want to log something? This might also work in a similar, the thread would then also write the values to a database or something.

I am sorry that I can't share any of the code I have running but there are some very good tutorials on multithreading with python out there!

Regards
Max

closed 2021-05-14 by drunsinn