drunsinn/pyLSV2 #7: Sending file in ASCII mode - error

kekec14, 2021-04-13:
Hello!

I own:
`iTNC530, NC Software 606420 02 SP14` machine

I am strugling with sending txt file (two blocks - more than 4096bytes) to the machine. If file is sent in binary mode everything is ok. When I try to send the same file in ASCII mode I recieve this error:

`ERROR:root:could not send data with error T_BD`

If file is small (only one block - up to 4096 bytes) both sending modes works ok.

Thank you!

drunsinn, 2021-04-13:
Hello @kekec14,

Can you post a bit more of the log? Do you have any specific reason that you can't use binary mode?

As far as I understand the difference between ASCII mode and binary mode is that for ASCII is the older one which had problems quite similar what you are experiencing. I would therefor recommend to always enable binary mode for all transfers. There doesn't seem to be any downside to transferring files in binary mode as far as I am aware, it has always worked just fine for me. In TNCremo you can also enable binary mode for all transfer operations.

Thanks for sending your software version, I will add it to the list of known working versions!

drunsinn, 2021-04-29:
@kekec14 did this answer help you out? If yes, could you close this issue?

closed 2021-04-29 by kekec14