drunsinn/pyLSV2 #8: Recive file in ASCII mode

kekec14, 2021-04-13:
HeLlo!

I own:
`iTNC530, NC Software 606420 02 SP14 machine`

I have another question. In the `client.py` > `def recive_file` :

```
with local_file.open('wb') as out_file:
            #file_buffer = bytearray()
            if response in self.RESPONSE_S_FL:
                # file_buffer.extend(content)
                if binary_mode:
                    out_file.write(content)
                else:
                    out_file.write(content.replace(b'\x00', b'\r\n'))
                logging.debug(
                    'received first block of file file %s', remote_path)

                while True:
                    response, content = self._llcom.telegram(
                        LSV2.RESPONSE_T_OK, payload=None, buffer_size=self._buffer_size)
                    if response in self.RESPONSE_S_FL:
                        # file_buffer.extend(content)
                        if binary_mode:
                            out_file.write(content)
                        else:
                            out_file.write(content.replace(b'\x00', b''))
                        logging.debug(
                            'received %d more bytes for file', len(content))
                    elif response in self.RESPONSE_T_FD:
                        logging.info('finished loading file')
                        break
                    else:
                        if response in self.RESPONSE_T_ER or response in self.RESPONSE_T_BD:
                            logging.error(
                                'an error occurred while loading the first block of data for file %s : %s', remote_path, self._decode_error(content))
                        else:
                            logging.error(
                                'something went wrong while receiving file data %s', remote_path)
                        return False
```
in `while True:` loop it should also be:
```                       
 else:
          out_file.write(content.replace(b'\x00', b'\r\n'))
```

Or am I wrong?

Thank you!

drunsinn, 2021-04-13:
@kekec14 you are correct! Only the bytes in the first block got converted back to a new line.
I will publish a new version with the fix shortly!

drunsinn, 2021-04-13:
Just pushed version 0.6.4 to pypi.org should be available for update shortly :-)