drunsinn/pyLSV2 #10: Tool Call

kekec14, 2021-05-12:
Hello!

Is it possible to get information which tool is currently in use? Sam principle as getting the info about current running program. 

`get_program_stack()`

Thank you!

drunsinn, 2021-05-12:
Hello @kekec14,

currently the library has no such function but I will take a look at it this weekend. The R_RI command has a lot of command types I have not yet decoded. I will let you know here I I find anything!

If this dosen't work out you could also load the pocket table from the control and check the row for the spindle. As long as you don't write to this file there shouldn't be a problem with 'loosing' any tools. For parsing the table files I have recently added a table parser.

kekec14, 2021-05-13:
Where can I find pocket table?

drunsinn, 2021-05-13:
Thank you, that makes things easy! An update should be ready during the weekend if I have the time.

The control keeps track of the position of all the tools in the file TNC:/table/tool_p.tch. Line 0.0 is the spindle.

kekec14, 2021-05-13:
I found something interesting:

R_RI code to tigger too linfo is 51 (hex 33).

drunsinn, 2021-05-14:
Hello @kekec14

Thanks for the code! I just published a new version with your code! I also added the other data field (length, radius, axis and index).
I added the following code to the demo script that uses the fallback method if runinfo x does not work:

```

    t_info = con.get_spindle_tool_status()
    if t_info is not False:
        print('Current tool number {Number}.{Index} with Axis {Axis} Length {Length} Radius {Radius}'.format(**t_info))
    else:
        print('Direct reading to current tool not supported for this control type. using backup strategy')
        if con.is_tnc():
            pocket_table_path = 'TNC:/table/tool_p.tch'
            transfer_binary = True
            spindel_lable = '0.0'
        elif con.is_itnc():
            pocket_table_path = 'TNC:/TOOL_P.TCH'
            transfer_binary = False
            spindel_lable = '0'
        else:
            pocket_table_path = 'TNC:/table/ToolAllo.tch'
            transfer_binary = True
            spindel_lable = '0.0'

        with tempfile.TemporaryDirectory(suffix=None, prefix='pyLSV2_') as tmp_dir_name:
            local_recive_path = Path(tmp_dir_name).joinpath('tool_p.tch')
            con.recive_file(local_path=str(local_recive_path), remote_path=pocket_table_path, binary_mode=transfer_binary)
            tr = pyLSV2.TableReader()
            pockets = tr.parse_table(local_recive_path)
            spindel = list(filter(lambda pocket: pocket['P'] == spindel_lable, pockets))[0]
            print('Current tool number {T} with name {TNAME}'.format(**spindel))
```

This basically reads the pocket table from the control and looks for the correct line. It might look a bit complicated since I made it so it also works on other controls like older iTNC530 (pre 34049x 04).

I would really like to know why they removed this function on the TNC640 :-)

kekec14, 2021-05-14:
Thank you for the implementation. It works as expected!

closed 2021-05-14 by kekec14