drunsinn/pyLSV2 #13: Get error code

kekec14, 2021-05-24:
Hello!

Is it possible to recieve error code? I tried to read plc marker 4177 with this code:
`print('PLC Marker PR Mode Error active: {}'.format(con.read_plc_memory(address=4177, mem_type=pyLSV2.PLC_MEM_TYPE_MARKER, count=1)))`
which is True if there is an error. I would like to know what is error reason..

Thank you!

drunsinn, 2021-05-24:
Hello @kekec14,

reading the error message seems to be possible via TNCcmd with the commands RUNINFO F and RUNINFO N. The documentation says that this command works with a TNC640 only if you have TNCremoplus which is a paid option.

For an iTNC530 it seems to work just fine. I have a pushed a working version to master so you can take a look. As far as I can see it works as expected for the most part. One open question is on how to decode the first four bytes of the message and how the map to the result to the error group and error class displayed by TNCcmd. This will need some more testing!

Bevor I publish a new version with this function I also want to change it so you only have to run it once to get all current error messages. This means the parameter "next_error" will probably disappear in the release and it will return a list of error messages.

kekec14, 2021-05-25:
I am aware of `TNCcmd runinfo F/N`option. As you said it only works on TNC530... I thought if there is universal option for all type of machines (maybe via PLC address). I also have TNCremoplus and I can confirm it gives outuput on TNC640, but I would like to achive this with python library - if possible.

drunsinn, 2021-05-25:
At the moment I have no idea if there is anything in the PLC memory like that. There might be but I am not aware of any universal address to geht error codes or messages.
Can you capture the Interaction between TNCremoplus and the controls with Wireshark? Depending on how RUNINFO F/N is enabled I might be able to do the same in python but I expect there to be some kind of encryption going on for these paid operations.
I will look through the available documentation on the internet but I can't promise anything.

drunsinn, 2021-05-28:
@kekec14 do you know the PLC key for the control? There is a special drive called LOG: where you can access log messages from different parts of the control. If you can find a file that contains the log message you are interested in it would be possible to create a file parser similar to the table parser

kekec14, 2021-05-29:
Sorry right now I can not provide wireshark log. How can I access LOG: partition?

Thank you!

drunsinn, 2021-05-29:
If you use the key number for the PLC in the MOD menu you can access this partition from the file manager on the controls. The default for this key number is 807667 but this does not work on all of our machine.
[image]
Natural you can also access it via TNCremo or pyLVS2 :-)

drunsinn, 2021-07-05:
@kekec14 did you make any progress on this matter? I never managed to get the encrypted functions from TNCremo plus to work.

kekec14, 2021-07-07:
Hello! I did not manage to decrypt payload. I did workaround, which execute tnccmdplus.exe runinfo f, if plc marker for error is True.
I have additional question regarding TNC640 marker functions. In technical manual I noticed that there are some words beside M PLC markers... For example: `NN_ChnErrorFStop  (M4220)`.

My question is if this marker can be read via NN_ChnErrorFStop  and not via M4220?

drunsinn, 2021-07-07:
Hello @kekec14,

just as i expected, thanks for the feedback.

Regarding the modern style of the plc markers I have some good and some bad news :-/ From what I understand the control can either use direct memory addresses or an API with names.
TNCcmd does have the command GETVALUE which can read the values with the API name like this.
`GETVALUE \PLC\program\symbol\global\ApiChn[0].NN_ChnErrorFStop`
Again, this is done via an encrypted(?) transmission which i could not decode.

The good news is that it is still possible to get the values via the `read_plc_memory` command. You 'only' have to first manually check on the control which address is used for which API name. The control has a debug mode for the PLC program, this contains a mode called WHATCHLIST. To get there, enter the PLC key number, select LISTS and WATCH LIST. There you can select API names from a symbol list and add them to the watch list. After adding the name to the table it adds the memory address and the memory type necessary for `read_plc_memory`. I found that on my programming station the address for `NN_ChnErrorFStop` wasn't 4220 but either 4308 or 4600.

[image]

I restarted the programming stations multiple times and this address never changed so i expect as long as the PLC program isn't changed the address is always the same. It's no solution which can easily be translated from one machine to another, no real machines I checked had similar address values as the programming station.

devac33, 2021-08-17:
We cannot get the error code on some tnc530. What can be wrong?

drunsinn, 2021-08-17:
@devac33 can you be a bit more specific? What software version is running on the control? Can you read other information like the current operating mode? Reading error messages and reading the current mode use a similar lsv2 command.
As discussed here I know that it only works on iTNC530 controls but as far as I know there is no limitation regarding the software versions on the iTNC.

On another note: It would be better I you could create a new issue for your problem as it is nor related this issue.

drunsinn, 2021-09-06:
I am closing this issue since there wasn't any more feedback. Feel free to create a new one

closed 2021-09-06 by drunsinn