drunsinn/pyLSV2 #23: C-Sharp call Python

NecoMai, 2022-03-15:
This pyLSV2 program works for me very well.  Thanks all.
However, when I call this python program in c-sharp, I get an exception.
IronPython version is [3.4.0-alpha1 released on 2021-04-19](https://ironpython.net/download/)
The message from “Microsoft.Dynamic.dll” is "'NoneType' object has no attribute 'close'".
The code as below:

using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using Microsoft.Scripting.Hosting;
using IronPython.Hosting;

namespace ConsoleApplication1
{
    class Program
    {
        private static string address = null;
        private ScriptRuntime pyRuntime = null;
        private ScriptEngine pyEngine = null;
        private ScriptSource source = null;
        private ICollection<string> paths = null;

        static void Main(string[] args)
        {
            Program p = new Program();
            p.UsePython();
        }

        public void UsePython()
        {
            address = "192.168.1.37";
            System.Console.WriteLine("Connecting to {0}", address);

            pyRuntime = Python.CreateRuntime();
            pyEngine = pyRuntime.GetEngine("python");
            source = pyEngine.CreateScriptSourceFromFile(@"E:\JC\JCTest.py");

            ScriptScope scope = pyEngine.CreateScope();

            paths = pyEngine.GetSearchPaths();
            paths.Add(@"D:\Python\Python38\Lib");
            paths.Add(@"D:\Python\Python38\Lib\site-packages");
            paths.Add(@"E:\JC");
            pyEngine.SetSearchPaths(paths);

            source.Execute(scope); //  “System.MissingMemberException” in Microsoft.Dynamic.dll 
            System.Console.ReadLine();
        }
    }
}

Can anybody help me? Thanks very much!

drunsinn, 2022-03-15:
Hello @NecoMai,

thanks for the positiv feedback!

I have no experience on how to call a python library from C# so I can#t give you any feedback on this directly.
The only place in pyLSV2 where a .close() function is called is inside the disconnect function to close the tcp socket. If the socket is None I would think that something went wrong while establishing the connections. This can be because of a multitude of ways. pyLSV2 dosen't really handle exceptions very well at the moment, everything gets dumped to stdout in the end. Do you have a way to look at the output that is created bevor the MissingMemberException occurs?

NecoMai, 2022-03-15:
Hi drunsinn

      I'm glad to receive your email.  Thanks.

      I have found any of "*.close" in the package of pyLSV2 and commented it, but it doesn't seem to work at all. 

      I found that the problem occurs whenever importing the pyLSV2 package.

      c-sharp call Python

      Install IronPython in VS "Nuget package management" . It has two version on Python 2.7 and Python 3.4 .

NecoMai, 2022-03-15:
[image]

# This is a sample Python script.
# Press Shift+F10 to execute it or replace it with your code.
# Press Double Shift to search everywhere for classes, files, tool windows, actions, and settings.

import pyLSV2.client

# Press the green button in the gutter to run the script.
if __name__ == '__main__':

    address = '192.168.1.37'
    print('Connecting to {}'.format(address))

    # con = pyLSV2.LSV2(address, port=19000, timeout=5, safe_mode=False)
    # con.connect()
    # print(con.get_versions())
    # print('PLC Word: {}'.format(con.read_plc_memory(address=368, mem_type=pyLSV2.MemoryType.WORD, count=1)))
    # con.disconnect()

drunsinn, 2022-03-15:
I am currently not set up to do testing on windows so I can only make some wild guesses.

First I would recommend using `import pyLSV2` instead of `import pyLSV2.client`. Importing the whole package implicitly loads the module __init__.py which makes sure everything in the library is accessible from your application/script.

Second your example should not run any 'active' code from pyLSV2. The close function call responsible for your problem might therefore be somewhere else...

It might sound stupid but have you checked if python script works on its own? Maybe there is a side effect to using IronPython that I am not aware of which breaks something in pyLSV2. I have not tested pyLSV2 with Python3.4 which IronPython3.4 is based on but I am not aware of any language features which might be incompatible.
What happens if you also comment the line it the import statement? Only leave and try running it again.
```
if __name__ == '__main__':
    address = '192.168.1.37'
    print('Connecting to {}'.format(address))
```

BTW:
I also just ran [vermin](https://github.com/netromdk/vermin) to check if there might be any problems, its output looks like this:
```
vermin -v .
Detecting python files..
Analyzing 26 files using 10 processes..
~2, ~3       <path>/pyLSV2/docs/conf.py
~2, ~3       <path>/pyLSV2/pyLSV2/__init__.py
!2, 3.4      <path>/pyLSV2/pyLSV2/client.py
!2, 3.4      <path>/pyLSV2/pyLSV2/const.py
2.6, 3.0     <path>/pyLSV2/pyLSV2/low_level_com.py
2.7, 3.0     <path>/pyLSV2/pyLSV2/misc.py
!2, 3.4      <path>/pyLSV2/pyLSV2/tab_test.py
!2, 3.6      <path>/pyLSV2/pyLSV2/table_reader.py
~2, ~3       <path>/pyLSV2/pyLSV2/translate_messages.py
2.7, 3.0     <path>/pyLSV2/scripts/check_for_LSV2_commands.py
!2, 3.4      <path>/pyLSV2/scripts/lsv2_demo.py
!2, 3.4      <path>/pyLSV2/scripts/lsv2cmd.py
2.7, 3.0     <path>/pyLSV2/scripts/ssh_tunnel_demo.py
!2, 3.6      <path>/pyLSV2/scripts/table_reader_demo.py
!2, 3.6      <path>/pyLSV2/scripts/test.py
~2, ~3       <path>/pyLSV2/setup.py
~2, ~3       <path>/pyLSV2/tests/__init__.py
2.4, 3.0     <path>/pyLSV2/tests/conftest.py
~2, ~3       <path>/pyLSV2/tests/test_connection.py
~2, ~3       <path>/pyLSV2/tests/test_file_functions.py
~2, ~3       <path>/pyLSV2/tests/test_keys.py
~2, ~3       <path>/pyLSV2/tests/test_machine_parameters.py
~2, ~3       <path>/pyLSV2/tests/test_plc_read.py
~2, ~3       <path>/pyLSV2/tests/test_read_info.py
!2, 3.4      <path>/pyLSV2/tests/test_transfer.py
~2, ~3       <path>/pyLSV2/tests/test_translation.py
```
The only part that requires anything higher than python3.4 is the table reader functionality which you are not using so you should be fine on that front. All the other stuff that require 3.6 is not part of the library itself but are part of the accompanying example scripts.

NecoMai, 2022-03-16:
[image]

NecoMai, 2022-03-16:
Thanks very much. Using python 3.4 and IronPython 3.4.0 can fit that issue.
I am sure that it may be something wrong in the library on the other version of python.

drunsinn, 2022-03-20:
@NecoMai 
So, I looked a bit deeper into ipython3.4.0a1 . I did not use it from inside a C# program but I ran the interpreter from the command line. There where indeed two problems in the code that prohibited it from running. After fixing both (see branch #25) I was able to run lsv2_demo.py whit 'regular' python and python and get the same output.

I wasn't able to reproduce the error you reported with the missing close command. I added a if statement to check for None so at least I should not raise the error.

It would be grate if you could check again with the new changes from #25 and give me some feedback.

drunsinn, 2022-03-28:
backport is part of release v0.7.4

closed 2022-03-28 by drunsinn

DuXiaoChuang, 2025-07-08:
> [image]

Heidenhain PLC Data|CNCnetPDM l7.0.0.4  Could you send me a copy of this software? thank you