drunsinn/pyLSV2 #24: How can I read the XYZ coordinates of the spindle

sazima, 2022-03-16:
I want to read the spindle position of heidenhan cnc, but I can't find function.

sazima, 2022-03-18:
Maybe so, but I'm not sure

```python
        payload = bytearray()
        payload.append(0)
        payload.append(0x16)
        result = self._send_recive_block(CMD.R_RI, RSP.S_RI, payload)

```

result: [bytearray(b'\x00\x050.0000\x000.0000\x00500.0000\x000.0000\x000.0000\x00X\x00Y\x00Z\x00B\x00C\x00')]

sazima, 2022-03-18:
> 

Hello, Could  you tell me how to download tnccmd? Thank you

drunsinn, 2022-03-18:
Great find!
I checkt with TNC 640 and iTNC programming station and both work with the command you found. It was a bit tricky to decode the result but it worked out in the end.
The second byte is the number of axes followed by the actual values followed by the axes letters in order. We are missing info about the first byte which is always zero but it dosen't seem to be necessary :-)

The new function returns a dictionary where the keys are the axes letters and the values are the numerical values.

Pull request #27 contains the implementation. If it works for you I will include it in the next release.

drunsinn, 2022-03-28:
function is part of release v0.7.4

sazima, 2022-04-09:
Test OK in Heidenhain 530

Valteris, 2023-03-30:
I used TNC dialog function SYSREAD