drunsinn/pyLSV2 #26: Machine PLC Restart on running LSV2_demo.py

SabarinathRajendran, 2022-03-18:
I was trying out the LSV2_demo script and below is my output, i have got T_ER from the PLC and also the PLC had issues and idea on how to connect with the iTNC530 machine to gets the progam_status execution_status and spindle speed and spindle feed data

```
Connecting to "IP of the machine"
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_WRONG_PARA
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_WRONG_PARA
Connected to "iTNC530" with NC Software "*****"
Running LSV2 Version 1 with Flags 255
Drive Info: [(removed for security reason)]
Current Folder TNC:/ Free Space: 2085650432 Attrib: ['*', '*', '*', '*']
PLC Marker: [False, False, False, False, False, False, False, False, False, False, False, False, False, False, False]
PLC Word: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
PLC Double Word: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
PLC String: [' ']
PLC Input: [False, True, False, True, True]
PLC Word Output: [0, 0, 0, 0, 0]
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\PLC\memory\D\7928". This does not work for all control types
PLC Double Word 7928: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\PLC\memory\K\1". This does not work for all control types
PLC Constant 1: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\PLC\memory\M\211". This does not work for all control types
PLC Marker 211: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\PLC\memory\B\7192". This does not work for all control types
PLC Byte 7192: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\PLC\memory\W\10908". This does not work for all control types
PLC Word 10908: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\PLC\memory\I\11". This does not work for all control types
PLC Input 11: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\PLC\memory\S\30". This does not work for all control types
PLC String 30: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\TABLE\TOOL\T\1\DOC". This does not work for all control types
DOC column of tool 1: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\TABLE\TOOL\T\1\L". This does not work for all control types
L column of tool 1: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\TABLE\TOOL\T\1\R". This does not work for all control types
R column of tool 1: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\TABLE\TOOL\T\1\PLC". This does not work for all control types
PLC column of tool 1: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\TABLE\TOOL\T\1\TMAT". This does not work for all control types
TMAT column of tool 1: None
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_NOT_FOUND
WARNING:root:an error occurred while querying data path "\TABLE\TOOL\T\1\LAST_USE". This does not work for all control types
LAST_USE column of tool 1: None
Current Language: 0
Current Program Stack: {'Line': 0, 'Main_PGM': 'TNC:/*****', 'Current_PGM': ''}
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_WRONG_PARA
WARNING:root:an error occurred while querying current tool information. This does not work for all control types
Direct reading to current tool not supported for this control type. using backup strategy
Traceback (most recent call last):
  File "lsv2_demo.py", line 104, in <module>
    con.recive_file(local_path=str(local_recive_path),
  File "/pyLSV2-master/pyLSV2/client.py", line 846, in recive_file
    response, content = self._llcom.telegram(CMD.R_FL, payload, buffer_size=self._buffer_size)
  File "/pyLSV2-master/pyLSV2/low_level_com.py", line 118, in telegram
    response_length = struct.unpack('!L', response[0:4])[
struct.error: unpack requires a buffer of 4 bytes
```

drunsinn, 2022-03-18:
Hello @SabarinathRajendran 
getting some errors is normal, this happens when the control dosen't support a certain command. I tried to make pyLSV2 be as forgiving as possible when it comes to these kind of problems. In your case it seems the control dosen't support reading PLC values.

If you could run the lsv2_demo script with logging level DEBUG it would also I can tell you exactly what didn't work.
There are differences in the available functions between iTNC530 and TNC640 and also between the software versions.

The only serious problem I see is the exception at the end. It happens at a place where it tries to read some parameter from the data received from the control. At this point it always assumed to be at least 4 bytes but from the message it seems to be less than that.
There is currently no debug message that would tell us what really gets received. Could you please add a line like
`print(response)`
just before the line where it gives the exception and post the output?

SabarinathRajendran, 2022-03-18:
@drunsinn thank you for the speedy response, will attach the debugging log once I run it again. It's seems to be an issues with set_sys_command function with this particular control, thank you for the speedy reply and I would also like to make a pull request where the warning can be made Bold 😅😅, those should be highlighted, thank you for the kind reply ,will keep you updated with the status and solutions.

drunsinn, 2022-03-18:
Pull requests are always welcome!

manusolve, 2022-06-11:
We are running a few iTNC 530 machines with 340422-08 on them. Using the demo script:
https://github.com/drunsinn/pyLSV2/blob/c85d1dc64ce7c5f7e2941d0f558a22a6c702798f/scripts/lsv2_demo.py

The control terminates the EXT process when trying to access the tool file. Seems like error above. Below is the output with logging.basicConfig(level=logging.DEBUG)

I added 
con.login(login=pyLSV2.Login.PLCDEBUG, password="807667")
after the block of read_plc_memory requests after line 36 to /lsv2_demo.py. The password is no big secret to the world.

Connecting to 192.168.1.32
DEBUG:root:Socket successfully created, host 192.168.1.32 was resolved to IP 192.168.1.32
INFO:root:safe mode is off, login and system commands are not restricted. Use with caution!
DEBUG:root:Connected to host 192.168.1.32 at port 19000
DEBUG:root:telegram to transmit: command CMD.A_LG payload length 8 bytes data: bytearray(b'\x00\x00\x00\x08A_LGINSPECT\x00')
INFO:root:login executed successfully for login Login.INSPECT
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x01')
DEBUG:root:command CMD.R_VR executed successfully, received S_VR with 8 bytes payload
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x02')
DEBUG:root:command CMD.R_VR executed successfully, received S_VR with 10 bytes payload
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x03')
DEBUG:root:command CMD.R_VR executed successfully, received S_VR with 11 bytes payload
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x04')
DEBUG:root:command CMD.R_VR executed successfully, received S_VR with 18 bytes payload
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x05')
DEBUG:root:command CMD.R_VR executed successfully, received S_VR with 24 bytes payload
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x06')
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: wrong parameter
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x07')
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: wrong parameter
DEBUG:root:got version info: {'Control': 'iTNC530', 'NC_Version': '340422 08', 'PLC_Version': 'GR34CB.00G', 'Options': '%0000111100000011', 'ID': '000.160.205.001.009.057', 'SPLC_Version': 'not supported'}
DEBUG:root:telegram to transmit: command CMD.R_PR payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00R_PR')
DEBUG:root:command CMD.R_PR executed successfully, received S_PR with 124 bytes payload
DEBUG:root:got system parameters: {'Marker_Start': 12064, 'Markers': 10000, 'Input_Start': 22064, 'Inputs': 1000, 'Output_Start': 23064, 'Outputs': 1000, 'Counter_Start': 24064, 'Counters': 144, 'Timer_Start': 24208, 'Timers': 1000, 'Word_Start': 64, 'Words': 10000, 'String_Start': 25208, 'Strings': 16, 'String_Length': 128, 'Input_Word_Start': 10064, 'Input Words': 1000, 'Output_Word_Start': 11064, 'Output_Words': 1000, 'LSV2_Version': 1, 'LSV2_Version_Flags': 255, 'Max_Block_Length': 4096, 'HDH_Bin_Version': 0, 'HDH_Bin_Revision': 0, 'ISO_Bin_Version': 0, 'ISO_Bin_Revision': 0, 'HardwareVersion': 0, 'LSV2_Version_Flags_Ex': 53, 'Max_Trace_Line': 0, 'Scope_Channels': 0, 'PW_Encryption_Key': 0}
INFO:root:setting connection settings for iTNC530 and block length 4096
DEBUG:root:use buffer size of 4096
DEBUG:root:telegram to transmit: command CMD.C_CC payload length 2 bytes data: bytearray(b'\x00\x00\x00\x02C_CC\x00\x07')
DEBUG:root:telegram to transmit: command CMD.C_CC payload length 2 bytes data: bytearray(b'\x00\x00\x00\x02C_CC\x00\x13')
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: wrong parameter
DEBUG:root:unknown or unsupported system command
WARNING:root:secure file transfer not supported? use fallback
DEBUG:root:telegram to transmit: command CMD.A_LG payload length 5 bytes data: bytearray(b'\x00\x00\x00\x05A_LGFILE\x00')
INFO:root:login executed successfully for login Login.FILETRANSFER
INFO:root:successfully configured connection parameters and basic logins. selected buffer size is 4096, use secure file send: False
DEBUG:root:version info already in memory, return previous values
Connected to "iTNC530" with NC Software "340422 08"
DEBUG:root:version info already in memory, return previous values
Running LSV2 Version 1 with Flags 255
DEBUG:root:telegram to transmit: command CMD.R_DI payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00R_DI')
DEBUG:root:command CMD.R_DI executed successfully, received S_DI with 170 bytes payload
DEBUG:root:successfully received directory information {'Free Size': 3590074368, 'Dir_Attributs': ['E', 'M', 'S', 'P', 'IN', 'DRV', 'DIR', 'H'], 'Attributes': (1, 131, 1, 39, 1, 16, 64, 8, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0), 'Path': 'TNC:/'}
Current Folder TNC:/ Free Space: 3590074368 Attrib: ['E', 'M', 'S', 'P', 'IN', 'DRV', 'DIR', 'H']

DEBUG:root:telegram to transmit: command CMD.A_LG payload length 9 bytes data: bytearray(b'\x00\x00\x00\tA_LGPLCDEBUG\x00')
INFO:root:login executed successfully for login Login.PLCDEBUG
[... cut ...]
  File "./demo-1.py", line 106, in <module>
    con.recive_file(local_path=str(local_recive_path), remote_path=pocket_table_path, binary_mode=transfer_binary)
  File "/odoo/.local/lib/python3.6/site-packages/pyLSV2/client.py", line 963, in recive_file
    CMD.R_FL, payload, buffer_size=self._buffer_size
  File "/odoo/.local/lib/python3.6/site-packages/pyLSV2/low_level_com.py", line 132, in telegram
    response_length = struct.unpack("!L", response[0:4])[
struct.error: unpack requires a buffer of 4 bytes

drunsinn, 2022-06-11:
Hello @manusolve,

yeah, the standard key numbers is quite an open secret if you can use any search engine :-)
I still decided to not include it in the code since I don't want to risk someone to ruin their machine by for example deleting files on the PLC partition. That is also the reason i added the safe_mode parameter.

Your problem is definitely the same bug as reported by @SabarinathRajendran.

I wasn't able to make any progress since i could not reproduce the same problem with the machines and programming stations I have access to.

A simple thing to try would be to force using binary mode for the file transfer. To do this change the block starting at line 153 in `lsv2_demo.py` from 
```
        elif con.is_itnc():
            pocket_table_path = "TNC:/TOOL_P.TCH"
            transfer_binary = False
            spindel_lable = "0"
```
to
```
        elif con.is_itnc():
            pocket_table_path = "TNC:/TOOL_P.TCH"
            transfer_binary = True
            spindel_lable = "0"
```
Sometimes the ASCII mode behaves a bit strange......

If this doesn't help you will have to do a bit more changes. To debug this further it would be great if you could modify the code in `low_level_com.py` so it prints exactly what the response looks like. To get more info what happens you will have to replace the block of code starting at line 131. Currently it should look something like this:
```
        if response is not None:
            response_length = struct.unpack("!L", response[0:4])[
                0
            ]  # read 4 bytes for response length
            response_command = response[4:8].decode(
                "utf-8", "ignore"
            )  # read 4 bytes for response type
        else:
            response_length = 0
            response_command = None
```

I would like you to change this block to this:

```
        if response is not None:
            logging.debug("received block of data with length %d and content %s" % (len(response), response))
            if len(response) >= 8:
                response_length = struct.unpack("!L", response[0:4])[
                    0
                ]  # read 4 bytes for response length
                response_command = response[4:8].decode(
                    "utf-8", "ignore"
                )  # read 4 bytes for response type
            else:
                # response is less than 8 bytes long which is not enough space for package length and response message!
                raise Exception()
        else:
            response_length = 0
            response_command = None
```
With this change all the received data is printed 1:1 as a debug message when you run the demo script again.

manusolve, 2022-06-12:
For our application we just need the current program name and the machine run state; this works. However thought the debug information would be of use. 

I'm running the master branch as of yesterday; (I skipped the binary/ascii mod you proposed); adjusted the low_level_com file as:

```
        if response is not None:
            logging.debug("received block of data with length %d and content %s" % (len(response), response))
            if len(response) >= 8:
                response_length = struct.unpack("!L", response[0:4])[
                    0
                ]  # read 4 bytes for response length
                response_command = response[4:8].decode(
                    "utf-8", "ignore"
                )  # read 4 bytes for response type
            else:
                # response is less than 8 bytes long which is not enough space for package length and response message!
                raise Exception()
        else:
            response_length = 0
            response_command = None
```

The machine control goes RED and halts. It says that a log file is in the SYS: partition. Cannot access that with TNCRemo. The password is not the same as the one for the PLC (nor can I see them in "iTNC 530 Technical Manual_02_2011_en" amongst others). 

Let me know any other tests you would like to make.

Here is the output.

```
DEBUG:root:login already active
DEBUG:root:telegram to transmit: command CMD.R_RI payload length 2 bytes data: bytearray(b'\x00\x00\x00\x02R_RI\x003')
DEBUG:root:received block of data with length 10 and content b'\x00\x00\x00\x02T_ER\x01\x18'
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_WRONG_PARA
WARNING:root:an error occurred while querying current tool information. This does not work for all control types
Direct reading to current tool not supported for this control type. using backup strategy
Getting tool with itnc method
DEBUG:root:telegram to transmit: command CMD.R_FI payload length 16 bytes data: bytearray(b'\x00\x00\x00\x10R_FITNC:/TOOL_P.TCH\x00')
DEBUG:root:received block of data with length 39 and content b'\x00\x00\x00\x1fS_FI\x00\x00\x06Xb\x9a(\xd8\x00\x00\x00\x00TNC:/TOOL_P.TCH\x00\x00\x00\x00'
DEBUG:root:command CMD.R_FI executed successfully, received S_FI with 31 bytes payload
DEBUG:root:successfully received file information {'Size': 1624, 'Timestamp': datetime.datetime(2022, 6, 3, 15, 29, 28), 'Attributes': 0, 'is_file': True, 'is_directory': False, 'is_write_protected': False, 'Name': 'TNC:/TOOL_P.TCH'}
DEBUG:root:loading file from TNC:/TOOL_P.TCH to /tmp/pyLSV2_sccxc0eg/tool_p.tch
DEBUG:root:telegram to transmit: command CMD.R_FL payload length 16 bytes data: bytearray(b'\x00\x00\x00\x10R_FLTNC:/TOOL_P.TCH\x00')
DEBUG:root:received block of data with length 0 and content b''
Traceback (most recent call last):
  File "./demo-1.py", line 106, in <module>
    con.recive_file(local_path=str(local_recive_path), remote_path=pocket_table_path, binary_mode=transfer_binary)
  File "/odoo/pyLSV2/pyLSV2/pyLSV2/client.py", line 963, in recive_file
    CMD.R_FL, payload, buffer_size=self._buffer_size
  File "/odoo/pyLSV2/pyLSV2/pyLSV2/low_level_com.py", line 142, in telegram
    raise Exception()
Exception
```

manusolve, 2022-06-12:
On another control with a similar later revision the result is the same; The tool file is called TOOL_P60.TCH on that machine.

`DEBUG:root:got version info: {'Control': 'iTNC530', 'NC_Version': '340422 12 SP3', 'PLC_Version': 'GR34CS.01G', 'Options': '%0000111100000000', 'ID': '000014581109', 'SPLC_Version': 'not supported'}`

```

DEBUG:root:login already active
DEBUG:root:telegram to transmit: command CMD.R_RI payload length 2 bytes data: bytearray(b'\x00\x00\x00\x02R_RI\x003')
DEBUG:root:received block of data with length 10 and content b'\x00\x00\x00\x02T_ER\x01\x18'
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_WRONG_PARA
WARNING:root:an error occurred while querying current tool information. This does not work for all control types
Direct reading to current tool not supported for this control type. using backup strategy
Getting tool with itnc method
DEBUG:root:telegram to transmit: command CMD.R_FI payload length 18 bytes data: bytearray(b'\x00\x00\x00\x12R_FITNC:/TOOL_P60.TCH\x00')
DEBUG:root:received block of data with length 41 and content b'\x00\x00\x00!S_FI\x00\x00\x0cpb\xa3oH\x00\x00\x00\x00TNC:/TOOL_P60.TCH\x00\x00\x00\x00'
DEBUG:root:command CMD.R_FI executed successfully, received S_FI with 33 bytes payload
DEBUG:root:successfully received file information {'Size': 3184, 'Timestamp': datetime.datetime(2022, 6, 10, 16, 20, 24), 'Attributes': 0, 'is_file': True, 'is_directory': False, 'is_write_protected': False, 'Name': 'TNC:/TOOL_P60.TCH'}
DEBUG:root:loading file from TNC:/TOOL_P60.TCH to /tmp/pyLSV2_0_025bqv/tool_p.tch
DEBUG:root:telegram to transmit: command CMD.R_FL payload length 18 bytes data: bytearray(b'\x00\x00\x00\x12R_FLTNC:/TOOL_P60.TCH\x00')
DEBUG:root:received block of data with length 0 and content b''
Traceback (most recent call last):
  File "./demo-1.py", line 107, in <module>
    con.recive_file(local_path=str(local_recive_path), remote_path=pocket_table_path, binary_mode=transfer_binary)
  File "/odoo/pyLSV2/pyLSV2/pyLSV2/client.py", line 963, in recive_file
    CMD.R_FL, payload, buffer_size=self._buffer_size
  File "/odoo/pyLSV2/pyLSV2/pyLSV2/low_level_com.py", line 142, in telegram
    raise Exception()
Exception
```

drunsinn, 2022-06-12:
Thanks for the new logs!

The password for the SYS drive seems to be a generated each day base on the date. That's at least what a service technician once told me.
You might be able to get at the log file if you create a service file. The service file contains a lot of different configuration and log files. Try looking through these files, maybe we are lucky and the lsv2 error log gets also saved to the service file.

Sadly, the only thing i was able to learn from them is that the transfer breaks at a point that is very hard to diagnose. The control seems to send a TCP package without any data right after the request for reading the file is received.
The branch I created contains the changes w talked about including so stuff that might cause issues but I am certain this wont fix anything.

I found a programming station for software 340494 08 SP2, with this the file transfer is also working. I haven't been able to figure out in which order these software version got release so I cant say for sure i found something similar to your versions.
With this programming station I was also able to capture the traffic with wireshark generated by TNCcmd for loading files from the control. The only difference i can make out is the fact that the official tools seem to prefer a smaller buffer size which causes the transfer to be split over two TCP package instead of one:

`tnccmd`
```
00000000  00 00 00 11 52 5f 46 4c  54 4e 43 3a 2f 74 6f 6f   ....R_FL TNC:/too
00000010  6c 5f 70 2e 74 63 68 00  01                        l_p.tch. .
    00000000  00 00 0b f7 53 5f 46 4c  0c 00 00 00 54 00 02 00   ....S_FL ....T...
    00000010  00 00 00 00 54 00 00 00  00 00 00 00 00 00 00 00   ....T... ........
    00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
<snip>
    00000BE0  00 00 00 00 00 00 ff ff  ff ff ff ff ff ff ff ff   ........ ........
    00000BF0  ff ff ff ff ff ff 00 00  00 00 00 00 00 00 00      ........ .......
00000019  00 00 00 00 54 5f 4f 4b                            ....T_OK 
    00000BFF  00 00 02 2d 53 5f 46 4c  00 00 00 00 00 00 00 00   ...-S_FL ........
    00000C0F  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
<snip>
    00000E0F  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
    00000E1F  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
    00000E2F  00 00 00 54 00                                     ...T.
00000021  00 00 00 00 54 5f 4f 4b                            ....T_OK 
    00000E34  00 00 00 00 54 5f 46 44                            ....T_FD 
```
`pylsv2`
```
00000000  00 00 00 11 52 5f 46 4c  54 4e 43 3a 2f 54 4f 4f   ....R_FL TNC:/TOO
00000010  4c 5f 50 2e 54 43 48 00  01                        L_P.TCH. .
    00000000  00 00 0e 24 53 5f 46 4c  0c 00 00 00 54 00 02 00   ...$S_FL ....T...
    00000010  00 00 00 00 54 00 00 00  00 00 00 00 00 00 00 00   ....T... ........
<snip>
    00000E10  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
    00000E20  00 00 00 00 00 00 00 00  00 00 54 00               ........ ..T.
00000019  00 00 00 00 54 5f 4f 4b                            ....T_OK 
    00000E2C  00 00 00 00 54 5f 46 44                            ....T_FD 
```

It would be a great help if you could do something similar by capturing a successful exchange between TNCcmd/TNCremo and the control with wireshark. Maybe there is a know but in the protocol, the official tools would have a work around based on the software version....

I also found no differences in the LSV2 status flags reported by your machine and my programming station.
```
'Control': 'iTNC530'
'NC_Version': '340422 08'
'LSV2_Version': 1
'LSV2_Version_Flags': 255
'LSV2_Version_Flags_Ex': 3837
```
```
'Control': 'iTNC530 Programm'
'NC_Version': '340494 08 SP2'
'LSV2_Version': 1
'LSV2_Version_Flags': 255
'LSV2_Version_Flags_Ex': 3837
```
For comparison this is what a TNC640 looks like:
```
'Control': 'TNC640'
'NC_Version': '340595 11 SP2'
'LSV2_Version': 1
'LSV2_Version_Flags': 159
'LSV2_Version_Flags_Ex': 988925
```

manusolve, 2022-06-12:
[remo-2.zip](https://github.com/drunsinn/pyLSV2/files/8886751/remo-2.zip)

Attached wireshark dump of tncremo transferring TOOL_P.TCH

drunsinn, 2022-06-13:
To be honest, I don't see much of a difference between your recording and what i would have expected. Maybe it really is the additional byte at the end of the R_FL command.

I updated the pull request #31 with a test script. It simplifies the connection and file transfer process a bit. Additional i added some `time.sleep(1)` to add some delay in case the control has a problem with keeping up with the transfer speed. It also adds the R_ST command before and after the file transfer. If you could try with this script it would be greatly appreciated!

The R_ST and S_ST commands are some kind of status request. I was never able to figure out what the response data is supposed to be. They are sent regularly but don't seem to do much if anything. Since having them would have meant using threads or something I decided to leave them out.

manusolve, 2022-06-15:
Once the a machine is free I'll make a test
Is it this branch -> 
[broken_file_transfer_itnc](https://github.com/drunsinn/pyLSV2/tree/broken_file_transfer_itnc)

drunsinn, 2022-06-15:
No hurry, i very much appreciate your help!
I also have to wait for my boss to give me the ok to test my code on the company's machines :-)

Yes, [broken_file_transfer_itnc](https://github.com/drunsinn/pyLSV2/tree/broken_file_transfer_itnc) is the branch with test code. The script I mentioned is in the folder scripts and is called `itnc_transfer_test.py`

manusolve, 2022-06-15:
No problem

Line 23:
`con = pyLSV2.LSV2("localhost" port=19000, timeout=5, safe_mode=False)`
to
`con = pyLSV2.LSV2(address, port=19000, timeout=5, safe_mode=False)`

```
./itnc_transfer_test.py 192.168.1.32
Connecting to 192.168.1.32
DEBUG:root:Socket successfully created, host 192.168.1.32 was resolved to IP 192.168.1.32
INFO:root:safe mode is off, login and system commands are not restricted. Use with caution!
DEBUG:root:Connected to host 192.168.1.32 at port 19000
DEBUG:root:telegram to transmit: command CMD.A_LG payload length 8 bytes data: bytearray(b'\x00\x00\x00\x08A_LGINSPECT\x00')
DEBUG:root:recived block of data with lenght 8
INFO:root:login executed successfully for login Login.INSPECT
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x01')
DEBUG:root:recived block of data with lenght 16
DEBUG:root:command CMD.R_VR executed successfully, received S_VR with 8 bytes payload
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x02')
DEBUG:root:recived block of data with lenght 18
DEBUG:root:command CMD.R_VR executed successfully, received S_VR with 10 bytes payload
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x03')
DEBUG:root:recived block of data with lenght 19
DEBUG:root:command CMD.R_VR executed successfully, received S_VR with 11 bytes payload
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x04')
DEBUG:root:recived block of data with lenght 26
DEBUG:root:command CMD.R_VR executed successfully, received S_VR with 18 bytes payload
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x05')
DEBUG:root:recived block of data with lenght 32
DEBUG:root:command CMD.R_VR executed successfully, received S_VR with 24 bytes payload
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x06')
DEBUG:root:recived block of data with lenght 10
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_WRONG_PARA
DEBUG:root:telegram to transmit: command CMD.R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x07')
DEBUG:root:recived block of data with lenght 10
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_WRONG_PARA
DEBUG:root:got version info: {'Control': 'iTNC530', 'NC_Version': '340422 08', 'PLC_Version': 'GR34CB.00G', 'Options': '%0000111100000011', 'ID': '000.160.205.001.009.057', 'SPLC_Version': 'not supported'}
DEBUG:root:telegram to transmit: command CMD.R_PR payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00R_PR')
DEBUG:root:recived block of data with lenght 132
DEBUG:root:command CMD.R_PR executed successfully, received S_PR with 124 bytes payload
DEBUG:root:got system parameters: {'Marker_Start': 12064, 'Markers': 10000, 'Input_Start': 22064, 'Inputs': 1000, 'Output_Start': 23064, 'Outputs': 1000, 'Counter_Start': 24064, 'Counters': 144, 'Timer_Start': 24208, 'Timers': 1000, 'Word_Start': 64, 'Words': 10000, 'String_Start': 25208, 'Strings': 16, 'String_Length': 128, 'Input_Word_Start': 10064, 'Input Words': 1000, 'Output_Word_Start': 11064, 'Output_Words': 1000, 'LSV2_Version': 1, 'LSV2_Version_Flags': 255, 'Max_Block_Length': 4096, 'HDH_Bin_Version': 0, 'HDH_Bin_Revision': 0, 'ISO_Bin_Version': 0, 'ISO_Bin_Revision': 0, 'HardwareVersion': 0, 'LSV2_Version_Flags_Ex': 53, 'Max_Trace_Line': 0, 'Scope_Channels': 0, 'PW_Encryption_Key': 0}
INFO:root:setting connection settings for iTNC530 and block length 4096
DEBUG:root:use buffer size of 4096
DEBUG:root:telegram to transmit: command CMD.C_CC payload length 2 bytes data: bytearray(b'\x00\x00\x00\x02C_CC\x00\x07')
DEBUG:root:recived block of data with lenght 8
DEBUG:root:telegram to transmit: command CMD.A_LG payload length 5 bytes data: bytearray(b'\x00\x00\x00\x05A_LGFILE\x00')
DEBUG:root:recived block of data with lenght 8
INFO:root:login executed successfully for login Login.FILETRANSFER
INFO:root:successfully configured connection parameters and basic logins. selected buffer size is 4096, use secure file send: False
DEBUG:root:telegram to transmit: command CMD.C_CC payload length 2 bytes data: bytearray(b'\x00\x00\x00\x02C_CC\x00\x05')
DEBUG:root:recived block of data with lenght 8
Connection uses buffer size 2048
DEBUG:root:telegram to transmit: command R_ST payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00R_ST')
DEBUG:root:recived block of data with lenght 20
DEBUG:root:command R_ST executed successfully, received S_ST with 12 bytes payload
response to R_ST was : bytearray(b"\x00\x00/ \x00\x00\'\x10\x00\x00V0")
DEBUG:root:telegram to transmit: command CMD.R_FL payload length 16 bytes data: bytearray(b'\x00\x00\x00\x10R_FLTNC:/TOOL_P.TCH\x00')
DEBUG:root:recived block of data with lenght 0
Traceback (most recent call last):
  File "./itnc_transfer_test.py", line 39, in <module>
    response, content = con._llcom.telegram(pyLSV2.CMD.R_FL, payload, buffer_size=con._buffer_size)
  File "/odoo/pyLSV2/pyLSV2/pyLSV2/low_level_com.py", line 141, in telegram
    raise Exception("response to short: %s" % response)
Exception: response to short: b''
```

drunsinn, 2022-06-17:
so nothing has changes :-(

I was hoping by recreating the same conditions it would work. Since it still dosen't work I would think it is something much earlier in the connection setup process. 

The only thing I could think of is adding a "chang_dicectory" bevor starting the file download. See the updated branch.

I am aware that this is probably using up quite some amount of your time but could I ask you to record network traffic of TNCremo from the point of the connection being established until the file transfer is done. Again, no hurry, whenever your time allows it.

manusolve, 2022-06-28:
Sorry for the delay. 
Attached is the wireshark of:
- Connect to machine with TNCRemo 3.2.404
- Open TOOL_T.TCH 

[i530-connect-and-transfer.zip](https://github.com/drunsinn/pyLSV2/files/9003912/i530-connect-and-transfer.zip)

drunsinn, 2022-06-29:
Thank you for the Wireshark file! There is nothing obvious different. I will do a deeper analysis and as a next step I will be to create a script that mimics the exact same message flow.

I don't know if I have time the next few days but I will let you know what I find as soon as possible!

drunsinn, 2022-07-03:
@manusolve I updated the script in branch #31 to match the communication as closely as possible. Here is the capture I got:

[mimic_itnc_traffic.zip](https://github.com/drunsinn/pyLSV2/files/9035076/mimic_itnc_traffic.zip)

I certainly see now where the differences are at the section where the connection is established but what exactly causes the reset on the control is still unknown.

manusolve, 2022-07-03:
Below is the debug from the new script.

For us the main use case of this is to get the current program and the run state of the machine. Which we can do already. If we can help improve the library then we are happy to help. Hopefully the doc you have in the other thread will show the way!

```
Connecting to 192.168.1.32
DEBUG:root:Socket successfully created, host 192.168.1.32 was resolved to IP 192.168.1.32
INFO:root:safe mode is off, login and system commands are not restricted. Use with caution!
DEBUG:root:Connected to host 192.168.1.32 at port 19000
DEBUG:root:telegram to transmit: command A_LG payload length 8 bytes data: bytearray(b'\x00\x00\x00\x08A_LGINSPECT\x00')
DEBUG:root:recived block of data with lenght 8
DEBUG:root:command A_LG executed successfully, received T_OK without any payload
DEBUG:root:telegram to transmit: command R_VR payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00R_VR')
DEBUG:root:recived block of data with lenght 55
DEBUG:root:command R_VR executed successfully, received S_VR with 47 bytes payload
DEBUG:root:telegram to transmit: command R_PR payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00R_PR')
DEBUG:root:recived block of data with lenght 132
DEBUG:root:command R_PR executed successfully, received S_PR with 124 bytes payload
DEBUG:root:telegram to transmit: command C_CC payload length 2 bytes data: bytearray(b'\x00\x00\x00\x02C_CC\x00\x03')
DEBUG:root:recived block of data with lenght 8
DEBUG:root:command C_CC executed successfully, received T_OK without any payload
DEBUG:root:telegram to transmit: command C_CC payload length 2 bytes data: bytearray(b'\x00\x00\x00\x02C_CC\x00\x06')
DEBUG:root:recived block of data with lenght 8
DEBUG:root:command C_CC executed successfully, received T_OK without any payload
DEBUG:root:telegram to transmit: command A_LG payload length 5 bytes data: bytearray(b'\x00\x00\x00\x05A_LGFILE\x00')
DEBUG:root:recived block of data with lenght 8
DEBUG:root:command A_LG executed successfully, received T_OK without any payload
DEBUG:root:telegram to transmit: command R_ST payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00R_ST')
DEBUG:root:recived block of data with lenght 20
DEBUG:root:command R_ST executed successfully, received S_ST with 12 bytes payload
DEBUG:root:telegram to transmit: command R_VR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_VR\x05')
DEBUG:root:recived block of data with lenght 32
DEBUG:root:command R_VR executed successfully, received S_VR with 24 bytes payload
DEBUG:root:telegram to transmit: command R_DI payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00R_DI')
DEBUG:root:recived block of data with lenght 178
DEBUG:root:command R_DI executed successfully, received S_DI with 170 bytes payload
DEBUG:root:telegram to transmit: command R_DR payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01R_DR\x01')
DEBUG:root:recived block of data with lenght 1460
DEBUG:root:waiting for more data to arrive, 1 bytes missing
DEBUG:root:telegram to transmit: command RSP.T_OK payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00T_OK')
DEBUG:root:recived block of data with lenght 646
DEBUG:root:telegram to transmit: command RSP.T_OK payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00T_OK')
DEBUG:root:recived block of data with lenght 8
DEBUG:root:telegram to transmit: command R_ST payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00R_ST')
DEBUG:root:recived block of data with lenght 20
DEBUG:root:command R_ST executed successfully, received S_ST with 12 bytes payload
DEBUG:root:telegram to transmit: command R_ST payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00R_ST')
DEBUG:root:recived block of data with lenght 20
DEBUG:root:command R_ST executed successfully, received S_ST with 12 bytes payload
DEBUG:root:telegram to transmit: command R_ST payload length 0 bytes data: bytearray(b'\x00\x00\x00\x00R_ST')
DEBUG:root:recived block of data with lenght 20
DEBUG:root:command R_ST executed successfully, received S_ST with 12 bytes payload
DEBUG:root:telegram to transmit: command CMD.R_FL payload length 16 bytes data: bytearray(b'\x00\x00\x00\x10R_FLTNC:/TOOL_P.TCH\x00')
DEBUG:root:recived block of data with lenght 0
Traceback (most recent call last):
  File "./itnc_transfer_test_2.py", line 89, in <module>
    response, content = con._llcom.telegram(pyLSV2.CMD.R_FL, payload, buffer_size=con._buffer_size)
  File "/odoo/pyLSV2/pyLSV2/pyLSV2/low_level_com.py", line 141, in telegram
    raise Exception("response to short: %s" % response)
Exception: response to short: b''
```

The ERR file on the control; I extracted manually from the HDD

```
HeROS386 V3.1T - FATAL ERROR MESSAGE
Sat Jun 25 18:37:54 2022
Time Stamp: 1656182274.89067
Page Fault
Error-Code = 00000005
ACTIVE TASK = EXT.T_LSVCV-2
EAX = 00168F24  
EBX = 00000001  
ECX = FFFFFFFF  
EDX = 00000054  
EDI = 00000001  
ESI = 00000001  
EBP = 01F9E9DC  
CS:EIP = 011B:02BB4F36  bin\ext.omf:CODE32:0X00083F36
SS:ESP = 0110:01F8B000
EFLAGS = 00013246
Fault Address: 00000001  
FPUSR = 0020  FPUCR = 0F7F
FPUIP = 00186056  
FPUDP = 01F97FBC  
    FPR0 = <Not in use>               FPR1 = <Not in use>           
    FPR2 = <Not in use>               FPR3 = <Not in use>           
    FPR4 = <Not in use>               FPR5 = <Not in use>           
    FPR6 = <Not in use>               FPR7 = <Not in use>           
Call stack:
02B4A801  bin\ext.omf:CODE32:0X00019801
02B48B1E  bin\ext.omf:CODE32:0X00017B1E
02B56A9A  bin\ext.omf:CODE32:0X00025A9A
02B3A61D  bin\ext.omf:CODE32:0X0000961D
02B3A53C  bin\ext.omf:CODE32:0X0000953C
02B3A4BA  bin\ext.omf:CODE32:0X000094BA
00182C8C  
```