drunsinn/pyLSV2 #29: Can I read the actual feed rate of spindle?

NecoMai, 2022-03-22:
Hi @drunsinn 
    I want to read the feed rate of spindle. I modify the program to read the PLC data from address 2 to 1000.

    Unfortunately,  I can see the actual feed rate on the device's screen, but nothing in plc data.

    I think the actual feed rate change frequently，so it doesn't write into the plc.

    How can I get that feed rate of spindle ?

    Thanks very much.

drunsinn, 2022-03-22:
Hello @NecoMai,

sorry but as far as I know there is no direct command to read the current feed rates, at least TNCcmd dosen't seem to have anything.

Finding the correct PLC address is a bit of a problem because you have to know the data format first bevor checking the address space. Otherwise it is pretty difficult.
You could try a different approach if you have a TNC640 or similar: Look for the symbol name of the value you need and use the function `read_data_path()` . This would look something like this:

```lsv2.read_data_path("/PLC/program/symbol/global/MG_BA_Automatik")```

If you have the passcode for your machine you can check the symbol name/path on the control by entering the PLC programming mode -> Lists -> Watchlist -> Symbol List.

NecoMai, 2022-03-22:
Hi drunsinn
        I had tried to ask the supplier how I can get my want. It told me that the password in the control will be changed everyday. 
  
        So I decided to find another way. Fortunately, I had found the open project pyLSV2. Thanks all of you.

        I found that the feed rate of spindle can be calculated this morning, so I want to read the tool's information. 

        I will have a try tomorrow morning. Thank you for your suggestion.

NecoMai, 2022-03-26:
Hi drunsinn
         I read the PLC address from 4 to 100000 step 4, and I had found that there is something wrong after PLC address 21480. 
         I searched the target value from the result, I found what my want finally. That is exciting.
         I had found the PLC address, and I can read the feed rate of spindle from PLC now. 
         There are something 'f' strings in table_reader.py on line 32, 44, 313, 316, 321, 351. 
         I had contacted with the engineer of heidenhain and he told me that I need to buy the interface of DNC.
         A pleasant exploration experience, thank you for all of you.

drunsinn, 2022-03-27:
Hello @NecoMai,
That is really good to know it worked and you can get the values you want! It's nice to know that this little project is of use to someone!

I understand the situation as follows: The controls support two protocols for communication: LSV2 and DNC. LSV2 is the older one as it can also be used via a RS232 connection. All commands are transferred as plain ASCII codes and binary data which can be decoded with some trial and error. It's main use is/was transferring files. Thanks to the work from [Eclipse-Plugin-Heidenhain](https://github.com/tfischer73/Eclipse-Plugin-Heidenhain) it was quite strait forward to get started with pyLSV2.
The newer protocol is DNC which offers a lot more functionality but also costs money to enable it on the controls and to by some development tools to write your own software. It is a higher level protocol which is a lot more difficult to reverse engineer.

The problem with the 'f' strings should also be fixed as soon as I merge pull request #25 which drops the required python version for everything to python3.4. Testing all the recent changes will take a bit more time but the new release will probably be ready soon.

NecoMai, 2022-04-26:
The issue may be closed.

closed 2022-04-26 by drunsinn

sazima, 2022-05-06:
```
self.read_plc_memory(368, MemoryType.DWORD)
```

DuXiaoChuang, 2025-04-29:
> Hi drunsinn I read the PLC address from 4 to 100000 step 4, and I had found that there is something wrong after PLC address 21480. I searched the target value from the result, I found what my want [...]

Hello, what's the address?

drunsinn, 2025-05-04:
@DuXiaoChuang the address depends on the PLC program of your machine. You will need to check all available addresses to find the one with the information you want.

DuXiaoChuang, 2025-05-04:
Thanks，how to  check all available addresses to find the one with the information

---- Replied Message ----
| From | ***@***.***> |
| Date | 05/04/2025 21:44 |
| To | ***@***.***> |
| Cc | Du ***@***.***>***@***.***> |
| Subject | Re: [drunsinn/pyLSV2] How can I read the actual feed rate of spindle? (Issue #29) |
drunsinn left a comment (drunsinn/pyLSV2#29)

@DuXiaoChuang the address depends on the PLC program of your machine. You will need to check all available addresses to find the one with the information you want.

patrickschultes, 2025-05-04:
Alternatively you can try the scope function, set the scope time done and do this in the intervall you need. It is not a pretty solution but it does the job without checking all plc addresses.

I am not sure if this is working on 640 but I do it in this way. 

Check the dictionary in the scope demo which signals are available

NecoMai, 2025-05-05:
I can't open the github. So I use the other email engine instead.

For example:

for num in range(2, 1000, 4):

    print(con.read_plc_memory(address=num, mem_type=pyLSV2.MemoryType.DWORD, count=1), end='')

    print(',', end='')

    if num % 80 == 0:

        print('')

By the way,  

The data memory addresses of different manufacturers are different, even for the same manufacturer, the data memory addresses of different devices are also different, and there is no fixed one. The memory address for storing data is determined by the device manufacturer during the first initialization. Some data may be mapped to multiple memory addresses.