drunsinn/pyLSV2 #32: Part Count Function for iTNC Machines

SabarinathRajendran, 2022-06-16:
Hello @drunsinn ,

is there a way to get the part count of the machine?.

drunsinn, 2022-06-19:
I was not aware something like this could be done, haven't heard about this "hidden" plc memory address before. Quite interesting, I will have to try this myself.

From reading this article I would say it should be as simple as reading the PLC address just like 
`counter = lsv2_connection.read_plc_memory(address=20, mem_type=MemoryType.WORD, count=1)`
Nothing fancy necessary I would think.

The way we solved this problem currently is to read/write the counter values from a table that is saved at a fixed location on the control.
Only recently they added a "official" parts counter in the way I mentioned in my first response. Here are some links to different solutions that use either a table or a QR parameter:
- [Workpiece counter with table](https://www.klartext-portal.com/en/tips/nc-solutions/solution/werkstueckzaehler-mit-tabelle/)
- [Workpiece counter with QR parameter](https://www.klartext-portal.com/en/tips/nc-solutions/solution/werkstueckzaehler-mit-qr-parameter/)

SabarinathRajendran, 2022-06-19:
Hello @drunsinn ,

Thanks, I was hoping for a default counter and as you have referred to the default counter i will try to look in to it and will share the function if i can take the value from it.

God speed.

closed 2023-03-29 by drunsinn