drunsinn/pyLSV2 #35: Few question

Baptou88, 2022-07-04:
Hi there,

I just discovered this lib, and I would be very interested in using it!

To try this lib I have on my computer "programming station itnc530 340 494 08 SP2"

However, I have a few questions:

first after the line con.connect() I have the following error:
`WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_WRONG_PARA`, is that there is documentation that explains the errors.

secondly, what password should we enter in the line `con.login(pyLSV2.Login.MONITOR,'password')`?
is it located in the file: PLC:\OEM.sys?

Baptou88, 2022-07-10:
Hello,
Thank you for your reply !

I just tried again without specifying a password, I get this message:

`INFO:root:login executed successfully for login Login.MONITOR
DEBUG:root:telegram to transmit: command CMD.C_LK payload length 1 bytes data: bytearray(b'\x00\x00\x00\x01C_LK\x00')
DEBUG:root:command CMD.C_LK executed successfully, received T_OK without any payload
DEBUG:root:command to unlock keyboard was successful
DEBUG:root:telegram to transmit: command CMD.C_EK payload length 2 bytes data: bytearray(b'\x00\x00\x00\x02C_EK\x01\xc0')
WARNING:root:T_ER or T_BD received, an error occurred during the execution of the last command: LSV2_ERROR_T_ER_FONT_NOT_DEFINED
WARNING:root:an error occurred while sending the key code 448`

drunsinn, 2022-07-10:
Ok, so the login worked but the button press is not recognized.

This is something that came to my attention just recently by reading the documentation provided in #34. It has a list of completely different key codes. The control you are communicating with probably expects one of these key codes. Why the error message is talking about fonts is a bit strange though.

I am currently working on incorporating the findings from this document into pyLSV2. In the meantime here are some key codes you could try:
```
MODE_MANUAL = 0x0048  # was 0x01C0
MODE_AUTOMATIC = 0x004B # was 0x01C3
MOD_DIALOG = 0x0042 # was 0x01C7

UPPER_A = 0x4150
UPPER_B = 0x4250
UPPER_C = 0x4350

NUMBER_0 = 0x3050
NUMBER_1 = 0x3150
NUMBER_2 = 0x3250
NUMBER_9 = 0x3950
```
If you could check if these work as expected I can go ahead an add them as a new enum with a second set of key codes!

Baptou88, 2022-07-22:
for some time now I have not been able to log in with `lsv2.login(login=pyLSV2.Login.MONITOR)`, even with freshly installed itnc 530

drunsinn, 2022-07-23:
This was caused by an mistake on my side. I accidentally removed the login MONITOR from the list of allowed logins in safe mode.

Version 0.7.6 with a fix should be available soon.

Baptou88, 2022-07-23:
thanks for the reply

whether with or without the "safe_mode" I systematically get an error

Baptou88, 2022-07-24:
I finally managed to connect, here is what I did:

first in the PLC:/oem.sys file I uncommented the line "REMOTE.LOCKSOFTKEYVISIBLE = YES"

secondly in the TNC:/tnc.sys file I uncommented the 3 lines beginning with "remote."

and third in PGM edit mode click "MOD" then set "External access" to true

Baptou88, 2022-07-24:
[image]

drunsinn, 2022-07-24:
Interesting, the exact same programming station works for me and I cant remember changing anything in *.sys files. 

I will add a FAQ section to the readme and list your steps for everyone to find.

BTW: which version of python are you using? I am working on a rewrite for a version 1.0, the minimal required version will most likely be 3.7 because I want to use dataclasses for all the dictionaries. The API will probably also change as it is quite a mess at the moment.

Baptou88, 2022-07-24:
I use Python 3.10.5

Baptou88, 2022-07-24:
https://www.inventcom.net/s1/_pdf/Heidenhain_TNC_Machine_Data.pdf

closed 2023-03-29 by drunsinn