drunsinn/pyLSV2 #41: Integrated Oscilloscope in iTNC530

Mohamed-ali7, 2022-12-06:
Hi,
I wanted to share this program with you.
This program simulates the Integrated Oscilloscope in iTNC530 to read all values like the Oscilloscope channels.
The link is here:
https://drive.google.com/drive/folders/1taZjLOv6cMAPJqlqVHBUiwzT6qpI0O7g?usp=sharing

I wonder if you can write a python code for it.
Thanks.

drunsinn, 2022-12-11:
Hello @Mohamed-ali7,

thanks for the software! I will and check it out and see if I can capture and decode the network traffic.
As soon as I have something I will get back to you.

Mohamed-ali7, 2022-12-18:
I wonder if you are facing any problems with the software or want me to check any of its functions on the CNC.

drunsinn, 2022-12-18:
Thanks for the offer but I think I have figured it out as far as it is necessary to record data. After some trial and error I was able to capture the traffic of reading values from a programming station.

Currently I am able to replay the exact same sequence sent by TNCscope and I get the messages exactly as they should be. The problem is that I have to figure out what all the messages beint sent actually mean.

So far I identified 5 steps for reading data from one channel:
- load a list of available channels from the control
- setup step 1 : send 4 bytes -> returns 8 bytes
- setup step 2: send 12 bytes -> returns 22 bytes
- setup step 3: send single command -> returns 4 bytes
- setup step 4: send 16 bytes
- repeatedly read data -> returns 138 bytes
- tell control to stop sending data

As you can imagine without any documentation it isn't easy to decode all the data sent back and forth.

If you want you can try yourself. I am working on a branch based on v1 and if you are interested I can publish the current state so you can also take a look.

Mohamed-ali7, 2022-12-18:
No problem.

[image]

You can find a help file installed automatically with the program.
A screenshot taken from it is shared above.

Mohamed-ali7, 2022-12-18:
I also have the LSV2 protocol help file, I can share it with you if you think it may help in decoding.

Mohamed-ali7, 2022-12-18:
[SDK_Help.zip](https://github.com/drunsinn/pyLSV2/files/10254868/SDK_Help.zip)

drunsinn, 2022-12-19:
The help files for TNCscope I already found myself, thank you. The problem is that it is about the usage of the tool itself and not about the details of implementing the necessary LSV2 protocol.

On the other hand the documentation on the LSV2 protocol you posted is quite interesting but it lacks anything regarding the messages used for reading data with the scope. By capturing the traffic I found the following messages: R_OC, R_CI, R_OP, R_DT and R_OD. Of these four, only R_DT is documented, the others are not.
This was probably done deliberately. The only option I see is to continue decoding the data by hand.

Again, if you want it would be a great help if you could help out. Maybe try capturing the network traffic with Wireshark yourself and check if you find any patterns.

Mohamed-ali7, 2022-12-19:
I will try to work on it as soon as I get free from the final exams.
But before doing so I will have to learn Wireshark first. So, it may take some time for me, but I will be happy to help.

Mohamed-ali7, 2022-12-19:
Maybe if you publish the current state of what you have done, I can take a look.

drunsinn, 2022-12-20:
Take a look at the branch https://github.com/drunsinn/pyLSV2/tree/scope, it contains the current state of what I was able to implement thus far:
- read channels (type, number, name and available signals)
- set reading interval
- select channels (hardcode)
- take x number of readings

I made a simple test script scripts/tst_scope.py which connects to the programming station, selects the first and second axes for channel 2 and three and takes 10 readings

Mohamed-ali7, 2023-01-04:
This software (free trial) can be much easier to decode its network.

We will use it directly for our research.

I hope you can add its functions in one of the following versions.

https://drive.google.com/drive/folders/1XEqJ70zC-oAksSEaK-njMtYWX-BC_cup?usp=sharing

Mohamed-ali7, 2023-01-04:
I hope everything is okay and that you are having some progress for TNCscope.

If it seems to be very difficult or impossible for decoding the TNCscope network, I can find similar softwares that can do the same functions to try to decode their network.

drunsinn, 2023-01-18:
Can you try the latest versions and check again? I made some updates to the code and the demo script. It now dumps the channel list and the recorded data to text files.

Mohamed-ali7, 2023-01-18:
Now I could connect, but it gives me :
    availible_signals = con.tst_read_scope_channels()
AttributeError: 'LSV2' object has no attribute 'tst_read_scope_channels'

I checked the attribute, it exists in LSV2 class in the client library.
What could be the reason?

[image]

Mohamed-ali7, 2023-01-18:
Yes, you were right. Thanks!
Now, the code works.
I will try to figure out the readings.
Thanks again.

Mohamed-ali7, 2023-01-23:
Hi, here I just wanted to share some parts of the C++ libraries used for reading some of the channel data. 
They refer to (R_CI, R_OP, and R_OD):
[lsv2_def.zip](https://github.com/drunsinn/pyLSV2/files/10478433/lsv2_def.zip)

Mohamed-ali7, 2023-01-23:
As far as I could understand, you could already get all the scope readings and decode some of them, while some are still not decoded.
For example, the data.txt file that you created already includes some final decoded values i.e. read as exactly reported in TNCscope software?
Right?
Like what reading you didn't decode?

I didn't understand why this raise keyword-shown in a circle- is put here.
[image]

I mean when it raised an Exception for some of the channel decoding.
When removed I get a value (in my case it was a value of zero, could it be wrong or should I remove it?)
One more important thing I noticed is the "intervall_us=" values.

Mohamed-ali7, 2023-01-23:
I could understand why we get 10 readings (we choose it in the code) . But in one reading, why do we get so many samples for the same signal?
Is it related to iteration over time? Then, what is the point of 10 readings?
[image]

drunsinn, 2023-01-23:
Hello,

first, thank you for the header files. I will check them and try to import as much as possible!

Regarding the other questions:
The basic data structures are decoded, correct. There are still a lot of places where the meaning of the values is not known. Most of the stuff ist just guesswork or copy&paste form the observed communication. For example decoding the channel description: The data contains 36 bytes of `0x00`. I don't know what they are used for. The same is true for decoding the recorded data. It contains a "header" of `0x00 0x20 0xff 0xff 0xff 0xff`. I have no idea what these values might indicate. The first two bytes might be an indicator of how many values are included in this package but I am not certain. Every recoding I have made always used the exact same header.
Another example of missing information is about the values returned from the control. Currently everything gets converted to an integer value. I am not sure if this is correct since I haven't done enough testing.

In general every place where you find `raise Exception(....)` is a situation where I just compare the current values with values from my recordings. If the code doesn't raise an exception it indicates that you are getting the exact same patterns.

The same goes for line 2121. This is just a "catch all" for situations with unexpected behavior during debugging. It's meant to make sure everything that is not as expected gets reported. If you get an exception at this place the control doesn't like something about how you set up the channels. This could be because of a lot of things. Take a look at the log messages the get a better idea. You can change how detailed the log messages are by changing the line `logging.basicConfig(level=logging.INFO)` in `tst_scope.py` to `logging.basicConfig(level=logging.DEBUG)`

I also observed that the interval value depends on the type of channel. This is consistent with the behavior in TNCscope, some channels require higher reading intervals than others.

BTW: All these `raise Exception()` will have to be replaced with proper error handling bevor it gets included in the official release.

The reason you get more than 10 values is also because this function isn't finished yet. It would be more correct to change it from 'number of readings' to 'number of TCP packages'. I will have to think about a way to correct this behavior in the future.

Mohamed-ali7, 2023-01-23:
Ok, thanks a lot. I will be testing these functions in the near future on some machining processes and comparing the readings to those from TNCscope. If I notice something important I will write to you.

Mohamed-ali7, 2023-01-29:
Hi,
here [https://github.com/Mohamed-ali7/pyLSV2/tree/scope](url) I will add anything new that I notice.
Currently, I add the channel signals as constants to be easier to call as shown below:
[image]

Also, I am considering creating a new function for real-time data readings. It will have the same structure as "tst_record_data" function except to open the user instantly. I still have to think about a good design for it since with the current structure I have each time to edit the client code file each time to get what I want. Though, I can now leak any data I want in real-time.

Here is the line from where we can get real-time readings:
[image]

Also, I noticed that each TCP package has 32 readings over time. For example, the sampling frequency is around 320 Hz:
So, to get readings for one complete second, you can set "num_readings = 10", which will give 320 samples collected over time with a frequency of approximately 320 Hz.

Md-aliya7, 2023-02-21:
Hi,
Do you think that calibrating the readings manually (as shown below) is the right way to get the readings as they appear in TNCscope software, or from the networks themselves they may include such a thing?
[image]

drunsinn, 2023-03-02:
Good progress!

My only concern would be with the mapping of the signal number to a name. I would expect this mapping to be dependent one the version of the control. A fixed mapping like in your case might not fit for a control with other signals. I would prefer another approach: Some code that tries to map the signal number to the names at runtime after reading the signal list.

If you think you have a working state you can send me a pull request so it gets integrated into the main version.

Md-aliya7, 2023-03-04:
I sent you a pull request (or that is what I think I have done :) ).
Can you check it?
All changes are in the branch (Scope) :
pyLSV2/client.py
pyLSV2/const.py
scripts/Real_Time_Readings

Md-aliya7, 2023-03-07:
Concerning the software I sent you recently, the iTNC530 software version (The versions I tried on are 606425 & 606424) of the CNC control is incompatible with the TNCopt version so, no connection takes place.
Do you know any suggestions for solving this problem?

Md-aliya7, 2023-03-07:
https://user-images.githubusercontent.com/119819019/223422926-13b39239-6bbc-4269-adc7-fb7aeaa6db69.mp4

Just to make sure, this is how I tried to connect.

drunsinn, 2023-03-07:
Ok, I just installed and tried this software but I also couldn't get it to work with a iTNC. They probably removed support for iTNC controls.
It did connect to a TNC 640 but I wasn't able to get any meaningful data. Capturing the traffic with Wireshark confirmed that it, just like TNCscope, works with some kind of login/encryption. :-/

Md-aliy7, 2023-03-14:
Hi,
The machine I use is KERN iTNC530
[image]

The len(data_set) in it was 98 instead of 106.
As a result, the decoded data I am getting (when I remove the raise exception command) is different from the real one.

Md-aliy7, 2023-03-14:
I modified the code (as shown below) to allow len(data_set) = 106 or 98.
I checked it and made sure it works correctly by comparing the reading with TNCscope readings. The data I checked was both current and position. I also used the factor and offset coefficients you have added recently. Thanks!
[image]

drunsinn, 2023-03-14:
Hello @Md-aliy7,

first, regarding your comment with connection to the TNC640: I was referring to the TNCopt software. I used it to connect to a programming station while capturing the TCP packages. Up to a certain point I could follow the data but then some encrypted password was used. This is similar/the same as when recoding the connection between TNCscope and a TNC640. Until we manage to reverse the  encryption I don't see any way to make it work.

second: Thank you for the update! I was just typing up my reply while I received it :-) The length check was just some sanity check since all packages I had seen so far had always a length go 106. Maybe it is a case of the number of axes being different from the programming station.
Since for type 1 and 4 the end is always 1 less than the total length it would be easier to just calculate it. I will make the change and update the branch!

For the factor and offset: I don't know if the calculation is correct. Can you check if the value has to be multiplied with the factor or if the offset has to be added first?

Md-aliy7, 2023-03-14:
I couldn't decide multiplication or addition should be done first because in (position, acceleration, current ) the added offset value is zero (as shown below).
I will check what other readings could have an offset value that is not zero.

[image]
[image]

That is a comparison of the actual position between pyLSV2 (right) and TNCscope (left) both on KERN iTNC530.
[image]

We also checked the compared the current readings between the two and graphed them.
pyLSV2 (right) and TNCscope (left) both on KERN iTNC530.
[image]

Especially in the current, some difference is noticed, mainly because we couldn't connect to the same machine using both software at the same time. Also, we only sampled each one for 15 seconds following each other.

drunsinn, 2023-03-14:
At least your test show that we get the correct values!
To be honest, I didn't expect we would be able to read any values at first. Looking at those graphs there dosen't seem to be much of a difference.
In the end, it isn't that important if the multiplication or the addition comes first. It is a simple change to switch them around if necessary.

drunsinn, 2023-03-19:
I checked and there is no direct timing info in the packages.
The function `get_time` is not directly related to the scope functions. It was used during the setup procedure of the signals. I decided to extract it to a separate function because it seems unnecessary. If you want to sync the time of the control with the start of the recording I would do something similar to this:
```
from datetime import datetime, timedelta
...
intervall = 3000
...
recording_start_time = datetime.now()
reading_counter = 0
for package in con.real_time_readings(selected_signals, duration, interval):
    signal_readings = package.get_data()
    readings_per_signal = len(signal_readings[0].data)
    for j in range(readings_per_signal):
        reading_time = recording_start_time + reading_counter * timedelta(microseconds=interval)
        .....
        reading_counter += 1
```
I haven't tested this code but it should work.

Md-aliy7, 2023-03-22:
Will you merge the scope branch with the master branch soon? or more work still to be done?

Md-aliy7, 2023-03-22:
It would be easier for my work colleagues to use the scope functions if they are integrated into the master branch.
Thank you!

drunsinn, 2023-03-22:
version 1.1 has just been released :-)
If it is ok with you, please close this ticket because the initial implementation is done. Feel free to create a new one if there is anything else!

closed 2023-03-22 by Md-aliy7

Md-aliy7, 2023-05-15:
Commenting on that quoted part in one of your replies:

> first, regarding your comment with connection to the TNC640: I was referring to the TNCopt software. I used it to connect to a programming station while capturing the TCP packages. Up to a certain [...]

Did you find any way to decrypt the data in TNC640? 
Could you try any AI assistance like Chatgpt, google bard, or Microsoft Bing AI? 

Probably one of them may help!