drunsinn/pyLSV2 #42: Exception: AttributeError: module 'pyLSV2' has no attribute 'xxyyzzxxyyzz'

ToM-MaS, 2022-12-28:
Hi :)
i'm just starting with pyLSV2. When I try to execute "lsv2_demo.py" i get following kind of errors. I tried to find where the mentioned attributes are defined but I did not have luck. I can't find where they are defined. Can anyone help?
Exception: AttributeError:
module 'pyLSV2' has no attribute 'PLC_MEM_TYPE_MARKER'
module 'pyLSV2' has no attribute 'PLC_MEM_TYPE_WORD'
module 'pyLSV2' has no attribute 'PLC_MEM_TYPE_DWORD'
module 'pyLSV2' has no attribute 'PLC_MEM_TYPE_STRING'
module 'pyLSV2' has no attribute 'PLC_MEM_TYPE_INPUT'
module 'pyLSV2' has no attribute 'PLC_MEM_TYPE_OUTPUT_WORD'
module 'pyLSV2' has no attribute 'LOGIN_MONITOR'
module 'pyLSV2' has no attribute 'KEY_MODE_MANUAL'

ToM-MaS, 2022-12-28:
Nevermind :D
I found the reason for my issue.  
The README.md   links to an old version of lsv2_demo.py in the section 

"See [lsv2_demo.py](https://github.com/drunsinn/pyLSV2/blob/c85d1dc64ce7c5f7e2941d0f558a22a6c702798f/scripts/lsv2_demo.py) for a demonstration of some of the functions."

drunsinn, 2022-12-29:
Hi,

thanks for the hint, I will try to find a solutions so the readme always refers to the most recent version!

commit eb37a53c (2022-12-29): Correct links in readme

fixes #42
commit 39916a3b (2022-12-29): Correct links in readme

fixes #42
closed 2022-12-29 by drunsinn