drunsinn/pyLSV2 #46: signals at long & short interval together

Md-aliy7, 2023-04-09:
Hi,

I noticed that TNCscope can read signals of long intervals (e.g., position and velocity: 3 ms) & signals of short intervals (e.g., current: 0.6 ms) together with considering the frequency of each one (like position is updated only one time when the current can be updated 5 times at interval = 600 us chosen for both of them. As shown in the attached picture (current & velocity at x-axis):
[image]

I could change the code to behave like as following:

1- Cancelling these two lines:
[image]

2- Cancelling these lines:
[image]

3- Adding this if condition (special for interval = 600 us):
[image]

while this (count_high_freq) increases by one at the end of each for-inner loop.
It gives the same exact behavior, but this is very specific for interval = 600 us (as mentioned above) because of the very soft code while TNCscope has fewer choices with more codes of conditions.
I wonder if you have a better solution for it.

Md-aliy7, 2023-04-15:
I suggest modifying the code to be:
[image]

What do you think?

drunsinn, 2023-04-15:
@Md-aliy7 sorry for the late reply.
I tried to follow your suggestions but to be honest I am not certain I understand all of it :-)
Nevertheless your changes look good to me, I see no problem to integrate them as they are.

Can you create a pull request? Would be easier for me to integrate.

drunsinn, 2023-04-15:
Your changes have been integrated and a new release should be available shortly, thanks for the code!

closed 2023-04-15 by drunsinn