drunsinn/pyLSV2 #49: Integrated Oscilloscope in TNC640

Md-aliy7, 2023-05-23:
Hi,

Commenting on that quoted part in one of your replies:

> first, regarding your comment with connection to the TNC640: I was referring to the TNCopt software. I used it to connect to a programming station while capturing the TCP packages. Up to a certain point I could follow the data but then some encrypted password was used. This is similar/the same as when recoding the connection between TNCscope and a TNC640. Until we manage to reverse the encryption I don't see any way to make it work.

Did you find any way to decrypt the data in TNC640? 
Could you try any AI assistance like Chatgpt, google bard, or Microsoft Bing AI? 
Probably one of them may help.

drunsinn, 2023-05-23:
@Md-aliy7 no, I haven't been able to make any progress in decoding the data. I will try your recommendation but I don't have any experience in using these LLM's. Still, it is worth a shot.
Will let you know if I make any progress.

Md-aliy7, 2023-05-25:
I am also trying to figure out how it was encrypted.
Here I share some packets for specific values (3000 & -3000) mm/min
and (5000 & -5000) mm/min.
The highlighted parts include these specific values:

[-specific_numbers.pdf](https://github.com/drunsinn/pyLSV2/files/11562918/-specific_numbers.pdf)

Md-aliy7, 2023-05-26:
From what I could notice,
each value reading consists from 16 bytes like (0xdc, 0x05, 0x00, 0x00, 0x00, 0x70, 
0xa7, 0x40).

The last two bytes represent the sign of the readings:
If it's positive, they become 0x40
If it's negative, they become 0xc0

Md-aliy7, 2023-05-26:
[Investigation.zip](https://github.com/drunsinn/pyLSV2/files/11580237/Investigation.zip)
This capturing is good for diverse situations of two variables (position and velocity in x axis).
It shows when one of them is fixed or changes, how the other changes.

Md-aliy7, 2023-05-28:
The SDK file among what I have sent to you before, I noticed this if it can help:
[image]

Md-aliy7, 2023-05-28:
[image]

From Chatgpt:

In the context of password encryption using PWEncryptionKey, the interpretation can be as follows:

peer0_33 (R_CI): This could represent a command or signal from the client (peer0) to the server (peer1) to initiate a client interaction. The string "R_CI" might stand for "Request Client Interaction."

peer1_33 (S_CI): This could be a response or acknowledgment from the server (peer1) to the client (peer0) regarding the client interaction request. The string "S_CI" might stand for "Signal Client Interaction."

peer0_34 (C_OP): This could represent a command or signal from the client (peer0) to the server (peer1) to perform an operation. The string "C_OP" might stand for "Client Operation."

peer1_34 (T_OK): This could be a response or acknowledgment from the server (peer1) to the client (peer0) indicating that the operation was successful. The string "T_OK" might stand for "Transaction Successful" or simply "OK."

peer0_35 (R_O ): This could represent a response or acknowledgment from the client (peer0) to the server (peer1) regarding an operation. The string "R_O" might stand for "Response Operation." **The additional characters after "R_O" could be encrypted password data.**

peer1_35 (S_O mm ): This could be a command or signal from the server (peer1) to the client (peer0) to perform an operation. The string "S_O" might stand for "Server Operation." The additional characters "mm" could represent encrypted password data or some parameter related to the operation.

peer0_36 (C_OP): This could represent a command or signal from the client (peer0) to the server (peer1) to perform an operation. The string "C_OP" might stand for "Client Operation."

peer1_36 (T_OK): This could be a response or acknowledgment

Md-aliy7, 2023-05-28:
In typical connection between TNC640 & TNCscope:
[image]

[image]

drunsinn, 2023-05-31:
Thanks for the data and your research so far. I also found some of the relations you mentioned.
From what I found, the encryption uses the key number, the current date and the random value read from the control. What I haven't been able to determine how these values are mixed together.

At least the encrypted data isn't compressed or anything. The package header (S_OD in this case) isn't encrypted and the length of the data also matches what is expected. The data on the other hand is encrypted.

drunsinn, 2023-06-17:
A professional will probably have more success than us poking around in the dark. You might also add that one part of the key is the date as well as a secret key sent by the control.

One thing that makes me a bit worried is the current story between Nintendo and the folks of the dolphin emulator. I don't want to get into any trouble with Heidenhain for including the encryption stuff directly into pyLSV2. I am trying to think of a way on how the actually en/decryption code is part of the user application while the library only offers some way to plug in these functions. I will think about how this might be achievable...