drunsinn/pyLSV2 #54: memory loss

Baptou88, 2023-10-14:
Hello everyone,

It's been a while since I've coded on this library, and I can no longer remember what the passwords are to connect, to access any information.
I would particularly like to connect with the "DNC" login to have the "execution_state" function

respectfully Baptou

drunsinn, 2023-10-16:
Hello @Baptou88,

are you talking about the PLC code number? The default that can also be found in the manuals online is 807667 but is might be that this has been changed on your machine.

Baptou88, 2023-10-17:
I finally managed to connect, the password is 807667 but I had to deactivate the safe Mode, Thanks!

drunsinn, 2023-10-20:
For monitor I have never needed a password, it always worked without one.

There are machine parameters to change the passwords. With the PLC password you should be able to check if there is something different.