drunsinn/pyLSV2 #55: an error occurred while querying system information

Andreas-strg, 2023-10-17:
Hello,

I only use Fletransver for my programming and don't need any DNC or monitor functions, but I still get error messages (iTNC530 and TNC640)
The data transfer then works as desired

Can I deactivate the other logins somehow or what could be the problem?

```
con = pyLSV2.LSV2("172.29.48.126")
con.connect()
con.login(login=pyLSV2.Login.FILETRANSFER)
```
Console:
```
received unexpected response RSP.T_BD
an error occurred while querying system information on turbo mode
received unexpected response RSP.T_BD
an error occurred while querying system information on dnc mode
an error occurred while querying system information on axes samling rate
```
Thanks alot

drunsinn, 2023-10-20:
These log messages are normal. During the setup phase all known info items are queried to check the capabilities of the control. The queries are handled by the same code as everything else, this is why you see these messages. There is no easy way to get rid of these log messages besides changing the log level of your application to "ERROR".

Andreas-strg, 2023-10-21:
D

closed 2023-10-21 by Andreas-strg