drunsinn/pyLSV2 #60: Announcement of the discontinuation of the LSV2 protocol

CuriousCrook, 2024-03-26:
Apologies if this enquiry is a little off-topic, but I suspect I've found the right specialists here.

I have been told that the LSV2 protocol has been discontinued and will no longer be available on newer CNCs in about 2-3 years. 
The successor is the RemoTools protocol, but that seems to be a crappy Microsoft COM+ component.

Then there is an [OPC UA server](https://www.heidenhain.de/opcua-nc-server). It seems really intuitive, but it is only available for newer cnc. With this [python opc ua toolkit](https://github.com/FreeOpcUa/opcua-asyncio) I have already made first experiences in controlling a heidenhain CNC via OPC UA.

As far as I understood, I have two options:
1. build a separate connection for old and new controls respectively:
    1a) old cnc = LSV2
    2b) new cnc = OPC UA
2. write a Python wrapper for the RemoTools COM component and connect all controllers via it.

Does anyone have experience of using com components in python?
I find it [incredibly complicated](https://stackoverflow.com/questions/5111799/understanding-how-to-use-com-in-python).

I am grateful for any tips.

drunsinn, 2024-03-27:
Hi @CuriousCrook!

Personally I would go for your option 1. It might be a bit more work during implementation to get the same data via both protocols but probably still less of a hassle than option 2. 

Another thing to consider is the OS you want your application to run on. If you want to support anything other than Windows, the COM solution won't work without something like [wine](https://www.winehq.org) which adds another layer of complexity. This was also the reason for me to focus on LSV2 since I needed a solution that can run on MacOS.

In the end I am also skeptical of the time frame for the final discontinuation of the LSV2 interface. The latest software version of the TNC7 still supports LSV2. The only "limitation" is the requirement to use a SSH tunnel for the connection. In earlier versions this was optional, now you have to change a machine parameter (see [manual](https://content.heidenhain.de/doku/tnc_guide/pdf_files/TNC7/81762x-18/tncguide/1369999_02_A_02.pdf) page 2318 under notes) to allow connections without SSH. I also think that there are too many applications like tool setters out there that only support LSV2 since it doesn't require a software option on the control. People would be very unhappy if the had to by a new tool setter because the bought a new machine.

Andreas-strg, 2024-06-19:
I thing OPC UA on Heidenhein is not realy, there are a lot of changes in the past and all Heidenhain tools from TNCremo and all the OEM tools still work with LSV2

OPC UA needs for every connection one licence and the setup is mutch more complex, i think the consumer are not happy with it...