drunsinn/pyLSV2 #62: OUTPUT_DWORD reading PLC memory?

LanderUrgoiti, 2024-05-23:
Hello:

I have been trying to get some values that we write in the PLC memory but I am unable to find them. 
I have loked in the TNC640 control and I can see that calling this function I get the correct values from the DWORD table:

`    AAA = con.read_plc_memory(first_element=27,mem_type=pyLSV2.MemoryType.DWORD, number_of_elements=1)`
`    print(AAA)`

The issue is that I am looking for a DWORD stored in another table ODWORD or something like that. I am guessing this means OUTPUT_DWORD since I can see the OUTPUT_WORD values correctly using this

`    AAA = con.read_plc_memory(first_element=27,mem_type=pyLSV2.MemoryType.OUTPUT_WORD, number_of_elements=1)`
`    print(AAA)`

My question is, Is thete a way to get OUTPUT_DWORD values

[image]
[image]

drunsinn, 2024-05-30:
I think I understand what you mean and hope that it is as easy as to add a new data type to the code based on the settings of the regular output words. Currently I can't disturb our machines so it would be grate if you could try out the changes I made in pull request #63 before I merge them.

drunsinn, 2024-06-18:
@LanderUrgoiti have you been able to confirm if the changes in #63 work?

drunsinn, 2024-06-24:
support for output d word was added with v1.4

closed 2024-06-24 by drunsinn