drunsinn/pyLSV2 #64: Can´t login to "MONITOR"

oxbown, 2024-06-11:
Hi!
I´m able to connect to the iTNC software simulator in localhost mode. I use most of the commands of the pyLSV2 library but I cant login to MONITOR user and cant use softkeys commands and so on...
My software versions are:

- iTNC530 - Software ver. 340494 05 SP6 
- PLC Ver: BASIS 54
- Python v.3.10.10

The code I use is:
```
import pyLSV2
con = pyLSV2.LSV2("127.0.0.1",safe_mode=False)
con.connect()
con.login(pyLSV2.Login.DSP) #login OK
con.login(pyLSV2.Login.MONITOR)
```

Return the code:
`error logging in as MONITOR`

I unlocked the "EXTERNAL ACCESS" in the MOD menu but i´ve no success.
[image]

oxbown, 2024-06-17:
Closing the virtual keyboard fixes the problem. Only one _MONITOR LOGIN_ can be established at the same time.

drunsinn, 2024-06-18:
Thank you for letting me know you were able to fix it yourself. The last view days I wasn't able to reproduce the behavior either on a machine or on a programming station since I always disable the keypad because of the screen size...
I will make sure to add a section to the manual about this!

closed 2024-06-18 by drunsinn