drunsinn/pyLSV2 #65: jh.Get() and jh.Set()

Andreas-strg, 2024-06-19:
Hello,

I wanted to ask very carefully if there is any way to use the read and write API of the more modern Heidenhain controlls, with them you could for example safely edit specific table columns without always overwriting the entire file

The API can be accessed in the NC program via **DATA READ** / **DATA WRITE**, in Python programs on the controller via **jh.Get()** / **jh.Set()** and externally via the TNCcmd.exe **GETVALUE** / **SETVALUE**.

The latest confirms to me that there could be a possibility here via LSV2 or DNC 

Example for a Heidenhain Python Line:
```
ToolRow = jh.Get( "\\TABLE\\TOOL\\T\\8\\\*" )
MotorName = jh.Get( "\\TABLE\\'PLC:\\table\\motor_oem.mot'\\NR\\10\\NAME" )
```

There is a Heidenhain Document "Python for Heidenahin Controls" there are the API specified, i can send it to you when you ned
This API would open up so many possibilities, but don't feel rushed :)

drunsinn, 2024-06-20:
Hello @Andreas-strg,

I have looked into supporting the *value commands for NCK TNC's but I had no luck so far.
As far as I can see, TNCcmd first tries to log in with "DATA" which fails. This seems to be a test if the control is an iTNC since those accept the DATA login. Afterwards TNCcmd uses the asks the user for the plc key. This key somehow encrypted and used to log in with FILEPLC. This encrypted key changes with time so I expect it to take the current time into account but I haven't been able to get any further than this.
In #49 someone mentioned about asking some professional to look into reversing the algorithm but I haven't heard any news on the topic :-/

Andreas-strg, 2024-06-24:
okay thank you, it is very interesting what heidenhain is doing here
maybe there will be another possibility in the future :)

drunsinn, 2024-06-24:
Yes, I hope someone more knowledgeable about cryptography can look into what is going on here.