drunsinn/pyLSV2 #66: FileTransfer returns ERROR

oxbown, 2024-06-20:
When I use the send_file I, got `could not send end of transmission telegram, and got the response 'RSP.T_ER'` but the file is sent OK to the controller...
So I found that in the original Eclipse-Plugin, the response for end of transmission is T_ER.

https://github.com/tfischer73/Eclipse-Plugin-Heidenhain/blob/03fb2beec0fb25970f94745700b8f3a8a002af48/Plugin/src/ch/tfischer/hh/client/LSV_Client.java#L65

This line https://github.com/drunsinn/pyLSV2/blob/9dafce0d71d0bd9f8a6eadddde69be77265f3cd4/pyLSV2/client.py#L1073
need to be changed to `if not self._send_recive(lc.RSP.T_FD, None, lc.RSP.T_ER):`

drunsinn, 2024-06-20:
Ugh, you are correct. This seems to be a copy&past error that happened during some refactoring word.
I will add a new test so this won't happen in the future. Thanks a lot! Expect a new release in one or two days.

drunsinn, 2024-06-20:
I looked a bit deeper into this, it is correct that there is an error but it isn't the wrong response code. 
Going by protocol documentation i was able to find, without secure file send, no response should be expected from the control. pyLSV2 was missing a check to suppress waiting for a response altogether!

PR #67 (2024-06-20): Fix broken file transfer because of missing check for non secure file transfer
closes #66
commit 281edd04 (2024-06-20): Fix broken file transfer because of missing check for non secure file transfer (#67)

* correct expected response code

fix #66

* add test with compatibility mode

force compatibility mode to disable secure file transfer

* add options for pytest to project config

* undo previous fix and add check for no expected response

* add test for inactive safe mode

* Update README.md

* update file formating
closed 2024-06-20 by drunsinn