drunsinn/pyLSV2 #70: Cant login on iTNC530E

David-Krizak, 2024-07-05:
However I try to connect to machine i get unknown or unsupported login
could not log in as user PLCDEBUG. 
Tried it on multiple machines.
code:

```
import pyLSV2

con = pyLSV2.LSV2("192.168.125.101")
con.connect()
print(con.versions.control)
print(con.read_plc_memory(32, pyLSV2.MemoryType.MARKER, 15))
print(con.execution_state())
con.disconnect()
```

console:
`iTNC530E
[]
5
unknown or unsupported login
could not log in as user PLCDEBUG
unknown or unsupported login
could not log in as user DNC
`
Am I doing something wrong?

drunsinn, 2024-07-06:
On which line do you get the error about the unsupported login?
If it is when trying to read from plc memory, it is because you need to disable the safe mode of pyLSV2. To do this, add `safe_mode=False` as a parameter to the line `con = pyLSV2.LSV2("192.168.125.101")`.
The safe mode limits the accepted login type. This is meant to stop you from damaging the machine.

David-Krizak, 2024-07-31:
it was my fault, found it. Sorry....

closed 2024-07-31 by David-Krizak