drunsinn/pyLSV2 #71: HH 640 - Leverage PLC Keynames to retrieve values

npalmerDNX, 2024-07-17:
I have been talking with the team at JOBs regarding some of their Heidenhains and they mentioned that some recent changes allow for getting/setting with a string key like 'ST_Workpiece_counter.DG_planner_workpieces'. They sent a sample program that runs on the controller and mentioned that the same information should be retrievable with the same key remotely.

`
M81
READ FROM PLC KEY " STG_workpiece_counter.DG_produced_workpieces " TO QR48;
;`

Have you seen or heard anything related to this during your investigations? I am trying to confirm the behavior now

drunsinn, 2024-07-18:
No, I haven't seen these address type used via LSV2 yet but I will look into it! Does your contact know it it is possible to read the value via TNCcmd?

I have at least seen the notation before though. When using TNCscope, one can load a map file `PLCProgramm.map`. After loading this file, it is possible to select PLC values by name from a list.
I stopped looking into reading PLC values this way because I encountered a number of places that require the encrypted login (see also #65) to enable the necessary access rights. 
- The map file is stored on the SYS partition which can only be accessed with login `FILESYS` and the encoded PLC password again
- Adding a named PLC value to the selected channels adds the login `OSZI_EX` to the setup procedure which again uses the encrypted password

The first obstacle is probably something minor since the map file can be acquired manually to get all the available names. Once those are known the file isn't necessary anymore.
Gettin access to the values is another story since the password is necessary. Going by the Wireshark capture, reading follows the same flow as the other values. I find it strange that reading an axis position value does not require the encryption but reading a simple PLC byte via its name does.

npalmerDNX, 2024-07-18:
I will ask about TNCcmd and let you know; they said that there were a number of changes for the 640s to "optimize" but nothing concrete.