drunsinn/pyLSV2 #72: Table header wrong format

LanderUrgoiti, 2024-08-01:
Hello, I found an error when using parse_table()
Some .TAB s I have in my TNC640 have a header like this for some reason I ignore
BEGIN [NAME].AFC.TAB MM
the '.AFC' before the'.TAB' was making the program to rise an error. 
I have fixed it by changing the table_reader.py script in the line 306. 
Original:
`                    r"^BEGIN (?P<name>[a-zA-Z_ 0-9]*)\.(?P<suffix>[A-Za-z0-9]{1,4})(?P<unit> MM| INCH)?(?: (Version|VERSION): \'Update:(?P<version>\d+\.\d+)(?: Date:(?P<date>\d{4}-\d{2}-\d{2}))?\')?(?P<mark> U)?$",
`
Modified:
`                    r"^BEGIN (?P<name>[a-zA-Z_ 0-9]*)\.(?P<extension>[A-Z]+)\.(?P<suffix>[A-Za-z0-9]{1,4})(?P<unit> MM| INCH)?(?: (Version|VERSION): \'Update:(?P<version>\d+\.\d+)(?: Date:(?P<date>\d{4}-\d{2}-\d{2}))?\')?(?P<mark> U)?$",
`

drunsinn, 2024-08-01:
good catch but I think you fix might break the regular case like
`BEGIN [NAME].TAB MM`
since it makes the extension mandatory.
Does your application require the extension and suffice to be available separately or can they be stored as one? If so, I think I can make the whole regex a bit simpler.

LanderUrgoiti, 2024-08-02:
Hello, I really don´t know. 
This extension is not in the name, you can only see it when you open the program with notepad
For example, the table name I can see in TNCremo
`KH37831_CORRECCIONES.TAB`
The header line of this table
`BEGIN KH37831_CORRECCIONES.AFC.TAB MM`
I will try to find out the meaning of this extension.