drunsinn/pyLSV2 #73: tnc 640，How to read the spindle speed, actual feed rate, and spindle load

wtnu200, 2024-12-16:
I checked the entire pyLSV2 and couldn’t find the method to read the spindle speed, actual feed rate, and spindle load. Please help me, thank you.

drunsinn, 2024-12-16:
Those values aren't available via the basic LSV2 API. On iTNC530 they can be read from PLC values but this option was removed for the TNC640.

I would recommend looking at the official DNC library from JH. DNC is much newer than LSV2 and has a lot more functionality.

David-Krizak, 2024-12-17:
What library you are talking about exactly? I have same problem.
At the moment im running tnccmd and getting values from there but its silly solution and its janky.

wtnu200, 2024-12-18:
> What library you are talking about exactly? I have same problem.
> At the moment im running tnccmd and getting values from there but its silly solution and its janky.

Thank you, but DNC is chargeable

wtnu200, 2024-12-18:
> Those values aren't available via the basic LSV2 API. On iTNC530 they can be read from PLC values but this option was removed for the TNC640.
> 
> I would recommend looking at the official DNC library from JH. DNC is much newer than LSV2 and has a lot more functionality.

thank you，But DNC ​​is charged。

wtnu200, 2024-12-18:
> What library you are talking about exactly? I have same problem. At the moment im running tnccmd and getting values from there but its silly solution and its janky.

pyLSV2？
Please tell me the command to get these data using tcncmd, thanks

drunsinn, 2024-12-21:
@David-Krizak I was referring to the library mentioned here: https://www.heidenhain.com/products/digital-shop-floor/connected-machining . As @wtnu200 mentioned, it isn't freely available. During the work on pyLSV2 we discovered that on TNC640 controls the connection is established via LSV2 but switches to DNC which messages are encrypted.

wtnu200, 2024-12-28:
I read some useful data through the function read_plc_memory.
tnc640:
Address 4593 spindle speed
Address 4594 spindle programming speed
Address 1112 feedrate

tnc530：
Address 92 spindle speed
Address 1262 feed rate

Thank you very much pylsv2

DuXiaoChuang, 2025-04-29:
> I read some useful data through the function read_plc_memory. tnc640: Address 4593 spindle speed Address 4594 spindle programming speed Address 1112 feedrate
> 
> tnc530： Address 92 spindle speed Address 1262 feed rate
> 
> Thank you very much pylsv2

Why can't I read the values using these addresses? The code I used is as follows:            
con.read_plc_memory(4593, pyLSV2.MemoryType.WORD, 2)

DuXiaoChuang, 2025-04-30:
> I read some useful data through the function read_plc_memory. tnc640: Address 4593 spindle speed Address 4594 spindle programming speed Address 1112 feedrate
> 
> tnc530： Address 92 spindle speed Address 1262 feed rate
> 
> Thank you very much pylsv2

What is the address of spindle load? Thanks