drunsinn/pyLSV2 #74: Contribution

patrickschultes, 2025-03-27:
Servus, 

I just want to ask if you are interested in small contributions. 
I have an additional DatasetLength which is valid for the scope signals and two additional iTNC530 software versions which are working on our side. 
Unfortunately I do not have the rights to open a new branch to send you a PR. 

Thanks
Regards 
Patrick

drunsinn, 2025-03-27:
Servus!

Yes, pull requests are always welcome! I have recived pull requests from others before so I think it isn't a setting on my end.
As far as I know you can only create a pull request for an existing fork of a repository. At least that is how it worked in the past. So first fork pyLSV2 so it appears in your repositories. On this fork you can create branches as you see fit. Based on one of the commits you make you can a pull request here. That is the same workflow I follow for this repository as well.
Maybe check the list of previous pull requests to get an idea how others did it in the past.

Looking forward to your contributions!

patrickschultes, 2025-04-02:
Servus, 

yeah.. my mistake, just to google for it were faster than opening an issue 😂
I added it in the readme as "contribution for dummies"

closed 2025-04-02 by patrickschultes